ratgdo/mqtt-ratgdo #2: Please publish the updated arduinoImprovBootstrapper sources

CrazyCoder, 2023-11-02:
It's not possible to compile the sources without hacks since https://github.com/PaulWieland/arduinoImprovBootstrapper/tree/ratgdo2.0 is not up to date with the current code.

Could you please publish the updated code you are using to produce the binaries? Thank you!

Original issue: https://github.com/PaulWieland/ratgdo/issues/166.

kmalinich, 2023-11-29:
+1 for this. Or +100.
I've already put a few hours into getting my ratgdo 2.5 hardware working with an ESP32 - and this seems to be a serious roadblock to compiling mqtt-ratgdo for ESP32. With the holidays approaching, it'll be a long time before I can even get my hands on an ESP8266 to run the compiled binaries. My original plan was to use an ESP32 with esphome-ratgdo, but since i have a secplus 1.0+ opener, I can't run the ESPHome version of the project, and so far it's been weeks and I haven't seen a single commit in the esphome-ratgdo repo pertaining to secplus 1.0+ support. If the updated/current arduinoImprovBootstrapper sources aren't published, I guess I'll have to shoddily port secplus 1.0+ from mqtt-ratgdo to esphome-ratgdo. Kinda really just let down by this whole project.

lukescott, 2023-12-08:
I would like some guidance on getting started, hence #10. I would like to contribute, but I want to make sure that the firmware I'm compiling matches the binary already installed so I can focus on adding to the project rather than chasing down issues that are already solved.

p-jean, 2023-12-16:
I can build my own RATGDO firmware by hacking some things, but it is incomplete.
Is there any reason not to release the whole source?

p-jean, 2023-12-23:
The gdoStateLoop() alternates between looking for a query (like 0x37, 0x38, 0x39, 0x3a..) and looking for the response to a query.
If the response is missing, such as the case of 0x37 mentioned above, then it can be handled nicely by adding a response-timeout.

Specifically, if RATGDO does not see a response to a query within about 30ms then it could give up on the response and start looking for a new query.

Currently, I believe that RATGDO never gives up on getting a response and it misinterprets whatever is seen next as a response, even if it come much later.

jdchaiken, 2024-07-12:
> I can build my own RATGDO firmware by hacking some things, but it is incomplete. Is there any reason not to release the whole source?

You have any luck here?  I want to modify the bootstrapper to try add Wireguard support so I can add my mom's Garage door without adding additional infrastructure.

kmalinich, 2024-07-12:
> > I can build my own RATGDO firmware by hacking some things, but it is incomplete. Is there any reason not to release the whole source?
> 
> You have any luck here?  I want to modify the bootstrapper to try add Wireguard support so I can add my mom's Garage door without adding additional infrastructure.

Try the esphome version of ratgdo. I was able to include it as a package and build locally from YAML. I added a few things specific to my setup. You could probably add Wireguard

jdchaiken, 2024-07-12:
She has a security 1.0 opener, and esphome doesn't work with it

FeatherKing, 2024-07-13:
> She has a security 1.0 opener, and esphome doesn't work with it

you can do sec 1 with esphome. here is my config

```
substitutions:
  name: ratgdov25i-30f19e
  friendly_name: ratgdo-hag-west
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

# Enable Home Assistant API
api:
  encryption:
    key: "Yl+.........."

ota:
  password: "72cb3....."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ratgdov25I-112233"

captive_portal:

web_server:
```

i had to flash minimal esphome and then flash this one because there isnt much memory

kmalinich, 2024-07-13:
I use it with my security 1.0+ opener