ratgdo/mqtt-ratgdo #5: Feature request / suggestion: autodiscovery topic.

hjdhjd, 2023-11-11:
First - thanks for your terrific work in providing a great solution for those in the myQ ecosystem. Absolutely phenomenal. I’m the author of the homebridge-myq plugin and I’m working on a similar solution for ratgdo. One thing I’d love to see that’s missing currently is autodiscovery. You’ve got something there for HomeAssistant, but I’d like to see something in MQTT that allows for autodiscovery so other solutions (e.g. mine, or RED, etc.) can more easily take advantage of it.

My suggestion/proposal, is to publish to a MQTT topic like `/home/garage/autodiscovery` or `/home/garage/heartbeat` with a payload of `devicename` at a regular heartbeat interval (say 10 seconds or 30 seconds). If something subscribes to that autodiscovery heartbeat, one can more easily autoconfigure things for the end user, and as a bonus, can also indicate to users when a device may be offline for one reason or another.

Ideally, the payload would be a JSON inclusive of the model number and serial number of the device, in addition to device name, but from my perusal of the code, it doesn’t look like you’ve found a way to determine device model and serial information.

PaulWieland, 2023-11-11:
I don't think a heartbeat is necessary because mqtt offers message retention. The availability of online and offline (using last will and testament) are already published.

A retained auto discovery topic with the device vitals is a great idea. The question becomes what path to publish it to. I'm thinking /ratgdo since there shouldn't be any collisions. 

Also FYI there is a native [HomeKit](https://github.com/ratgdo/homekit-ratgdo) version being developed. Not saying a homebridge version isn't useful but it might be a redundant effort. If you do want to continue making a homebridge version, would you mind managing the repo under https://github.com/ratgdo/homebridge-ratgdo ?

hjdhjd, 2023-11-11:
Not every MQTT broker manages state. Relying on the broker, especially in simpler environments, for history isn’t guaranteed to always do the right thing. Brokers can be swapped out, be ephemeral, or the user might have configured history to not be retained. TL;DR: broker history isn’t a reliable solution.

I’m watching the native HomeKit version being developed with interest and hope it’s a terrific integration and addition to the ecosystem. As to my own codebase - I appreciate the offer, but I’m going to maintain my own for the time being, though I’m open to it in the future. I manage and maintain several popular Homebridge plugins and I prefer maintaining them myself.

MYeager1967, 2024-01-20:
Paul, is there a last will and testament? I configured my RATGDO and took it offline until I could install it. After 4 days, it still showed online even though it was sitting on my desk unpowered....