ratgdo/mqtt-ratgdo #6: Feature request/suggestion: status command.

hjdhjd, 2023-11-11:
Another suggestion: it would be helpful if an additional command could be introduced: `/command/status`. When triggered with a payload of `true`, it would force an immediate publish of the current state across all the other status topics (availability, door, light, lock, and obstruction) such as `/status/door`. This is useful for applications/plugins that want to bootstrap state. 

At the moment, it doesn’t seem that ratgdo publishes state on a regular basis. State is only published in response to a command and its side effects.

Thanks again for a terrific solution.

PaulWieland, 2023-11-11:
There is already a query command available.
A future release will add support for non sec+ 2.0 protocols.

The states are all retained by the mqtt broker, there is no reason to publish them on an interval.

hjdhjd, 2023-11-11:
Totally missed this one - my bad.

And as I stated in the other issue…not all brokers are created (or configured) equally. Relying on them to do so isn’t an assumption I’m prepared to make. 😄

PaulWieland, 2023-11-11:
Ehh I'm on the fence about that. If you have an mqtt broker that doesn't support retain correctly then you better use a different broker? It's a fundamental feature of mqtt...

hjdhjd, 2023-11-26:
Reopening this issue...the current query command doesn't actually publish all the current states - it sets everything to unknown and waits for the next event (whenever that may occur) to publish a state update. What I'd really like to see is a true query command that forces a publish of all known state information to MQTT. Any thoughts on that?

Fail-Safe, 2023-12-26:
@hjdhjd I'm glad I stumbled upon this issue. I have been hunting the same type of functionality where a query command could force the re-publishing of current ratgdo state.

PaulWieland, 2023-12-26:
This is what the query command does (and what it has always done):
[image]

closed 2023-12-26 by PaulWieland

Fail-Safe, 2023-12-26:
> This is what the query command does (and what it has always done)

@PaulWieland Sorry I wasn't more clear in my post a bit ago. I have been trying unsuccessfully to get the query command to work. Given the examples you have here: https://paulwieland.github.io/ratgdo/01_features.html

Where am I going wrong?

If:
Device Name = "MyGarageDoor"
mqtt Prefix = "home/garage"

Then:
mqtt.topic = "home/garage/MyGarageDoor/command"
mqtt.payload = "query";

Is this not the proper way to query for all current statuses?

hjdhjd, 2023-12-26:
> This is what the query command does (and what it has always done): <img alt="image" width="313" src="https://private-user-images.githubusercontent.com/4663918/292927892-7fbf3a17-c039-4af7-9fa8-43c43dc073ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MzQyOTksIm5iZiI6MTcwMzYzMzk5OSwicGF0aCI6Ii80NjYzOTE4LzI5MjkyNzg5Mi03ZmJmM2ExNy1jMDM5LTRhZjctOWZhOC00M2M0M2RjMDczYWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjZUMjMzOTU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjBjYWM1N2RjYmQxNTVkZjNkNTM3NGNiNzBlNjk1NWI0ZTk3MmVlZmZlNDQ5ODhiMzUyMjQwM2RkZTVhYjEzZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pOxKKQszirA0dkBVCxetToJqNcRFCrHD2fyNZkQUYT0">

Yes and that’s part of the problem…it sets all the states to unknown and isn’t actually reporting on what ratgdo believes the current state is. What **should** happen is a publish of the current state.

Fail-Safe, 2023-12-26:
It's a little hard for me to interpret what is going on in the screen shot. It looks like some command (trying to get clarity on the proper syntax) was run which does reset all states to `unknown` but then a second later current statuses were published.

If this is the behavior, this would work fine for my use-case. I just can't seem to get the query command to work.

hjdhjd, 2023-12-26:
> It's a little hard for me to interpret what is going on in the screen shot. It looks like some command (trying to get clarity on the proper syntax) was run which does reset all states to `unknown` [...]

That’s essentially what it does…except what it should do is force a one-time republishing of state across the board.

Anything that’s listening for state changes will get an unnecessary (and incorrect) message pushed with an unknown state message.

Fail-Safe, 2023-12-26:
Does the query command work for you? If so, what syntax do you use?

Fail-Safe, 2023-12-27:
> This is what the query command does (and what it has always done): <img alt="image" width="313" [...]

Is it possible this behavior is different for those, like in my case, that are using the dry contact mode of operation?

I see where all states are set to unknown for a `query` command, per @hjdhjd's callout:
https://github.com/ratgdo/mqtt-ratgdo/blob/2.5/src/ratgdo.cpp#L554-L562

Is it possible that dry contact doesn't get an update after those reset states until some other event occurs? (Door open/close, light toggle, etc)

Goose66, 2024-01-31:
While I realize this issue is closed, I just wanted to add a +1 for getting the query command to work for drycontact mode.