ratgdo/mqtt-ratgdo #7: Make logs accessible thru web server

lukescott, 2023-11-19:
It's a big awkward standing on a ladder w/ the Ratgdo plugged into a laptop. Would be safer to be able to access the logs remotely.

dbrewer-wdc, 2023-11-20:
I do a lot of my debugging using the MQTT Explorer. But I guess it depends on what you want to do. MQTT explorer is a free download and shows all the commands and status sent.

lukescott, 2023-11-20:
@dbrewer-wdc Can you get logs thru the MQTT Explorer? I'm only seeing config. I'm using the Sec 1.0 beta and I'll need to provide some logs. As far as I know you can only get the logs by directly plugging into the USB port.

PaulWieland, 2023-11-20:
its on my todo list, but unfortunately kind of challenge to implement due to the underlying libraries used.