ratgdo/mqtt-ratgdo #9: Repurpose open/close/light terminals for reed switch?

lukescott, 2023-11-20:
I know the Ratgdo knows the door open/close state when in operation. What isn't clear to me is if it knows the door state when it is manually opened/closed (forced open).

The Ratgdo has triggers for open/close/light used to trigger these actions without needing WiFi. Could there could be an option to repurpose one of these terminals to read the state of a reed switch? And preferably report it as a different sensor. That way if the door was forced open the Ratgdo could report on it.

Previously I used a Z-Wave MIMI Lite w/ the reed switch to open / close the door. I could potentially keep the sensor plugged into that still, but I'm short an outlet. And it would be cleaner if the Ratgdo had a way to utilize the switch.

dbrewer-wdc, 2023-11-21:
I'm not a programmer by profession but I can get around the code ok. I've used Arduino IDE in the past and I have this project compiling on that platform for now. I have a goal to get PlatformIO working as well as VS Code, which is an editing environment that has a PlatformIO plugin. The reason I stuck with Arduino IDE for now is that it will download the compiled code to the ratgdo or ESP device. It will even do it over the air. And if attached serially, as soon as it is done loading, it starts up and you have your attached serial port. With that said, there were many pittfalls. In the platform.ini file included in this project, it will tell you which versions of the libraries to use as well as some compile time options that I had to hardcode into the code and libraries...not the best way to go. But if you know your way around a programming environment, PlatformIO or VS Code might be the way to go.

As for what you want to do, sure, that would be pretty simple. I don't think you need an OPEN and CLOSE button. I changed mine to just TOGGLE for the switches; however, for MQTT commands, I still use open and close. Seems more appropriate for automation. And now I use the CLOSE input for another feature. But yeah, a reed switch input would be perfect.

BR

lukescott, 2023-11-21:
@dbrewer-wdc I opened an issue [here](https://github.com/ratgdo/mqtt-ratgdo/issues/10) asking for a contribution doc. So you would say compiling and updating the firmware is fairly straight forward? I just want to make sure I don't end up bricking it.

I am a programmer for my day job, but I'm more on the web software / server end. I haven't done anything with Arduino.

dbrewer-wdc, 2023-11-22:
@lukescott. I don't know how you can brick it, but I'm sure it is possible. The ESP device is pretty cheap...it is a D1 Mini and you can get them on Amazon. However, unless you do something drastic to it, I think there is little chance of bricking it. With the Arduino IDE, which I motioned in the other post, there are quite a few gotchas to getting it to compile correctly, but if you look at the platform.ini file, it tells you which libraries to use and some compile options you'll need. The Arduino IDE can auto download the image after the compile. However, there are a few things you have to do to set it up. I'm sure PlatformIO is better and probably does all the same things...but actually reads the platform.ini file. I'd suggest finding a tutorial on YouTube programming a D1 Mini with PlatfomIO. Yup, there is. I searched, "d1 mini with platformio" on YouTube and a got quite a few short videos. I didn't watch any, but I will soon. Right now, I have it compiling so I'm just sticking with Arduino IDE for the remainder of this project.

BR

CrazyCoder, 2023-12-06:
The firmware has support for open/close reed switches, which work in dry contact mode for stupid GDO without security. I'm using exactly this setup for my Genie GDO, and it can detect a manual door open and close, see my writeup here: https://github.com/Kaldek/rat-ratgdo/issues/5.

dbrewer-wdc, 2023-12-08:
@PaulWieland that sounds reasonable. However, it does seem there are a number of doors like mine where the wall switch does not request the door status (38). Having a reed switch would be the best alternative for this scenario. Perhaps just assume reed switch mode for the door input upon power up unless the input is toggled a few times in rapid succession indicating a button is present....or some other variation.

I'd have offered up that code, but I currently use another ESP32 that monitors the door through a reed switch and issues a doorstatus MQTT command to the ratgdo (modified code with a new doorstatus command) so it can report to home assistant the correct door status. And just one reed switch is sufficient. 

For my opener, a 40 command comes back with 80 from the opener when the door is moving and 00 when it isn't. So if you wanted to know open closed opening or closing, that still could be determined, but I think just open/close is enough.

lukescott, 2023-12-08:
I don't even need the reed switch to be the door status. If it could report as an independent sensor / device that would go a long way for me in using it as a security sensor.

dbrewer-wdc, 2023-12-08:
@lukescott , yes, you could do that quite simply. But that would definitely be your custom code. You would need to create a new sensor status and then a report status via MQTT. If you have Homeassistant, you can even modify the home assistant code to report a new boolean sensor. Were you able to get the environment to compile?

PaulWieland, 2023-12-08:
> @PaulWieland that sounds reasonable. However, it does seem there are a number of doors like mine where the wall switch does not request the door status (38). Having a reed switch would be the best [...]

If you want to use reed switches, then just change your control protocol in ratgdo to dry contact mode. The reed switches are then wired to the trigger door inputs. ([see wiring diagram](https://paulwieland.github.io/ratgdo/03_wiring.html))