ratgdo/mqtt-ratgdo #11: GDO open/close and switching remote unlock/lock when ratgdo is rebooted (Sec+ 2.0 LM8355 + 882LM)

monkitrainer, 2023-11-24:
GDO = LiftMaster 8355 (Sec+ 2.0)
Wall Control Panel = 882LM
ratgdo HW 2.5 + 2.51 MQTT firmware

Issue:
GDO operates the door and wireless remote lock/unlock when the ratgdo is rebooted.

Steps to reproduce:

**GDO initial state = closed/unlocked**
Reboot ratgdo
_GDO opens and the wireless remote gets locked_

log: [ratgdo-open_lock_when_rebooted.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13461456/ratgdo-open_lock_when_rebooted.txt)

**GDO initial state = opened/locked**
Reboot ratgdo
_GDO closes and the wireless remote gets unlocked_

log: [ratgdo-close_unlock_when_rebooted.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13461455/ratgdo-close_unlock_when_rebooted.txt)

* ratgdo is working perfectly other than the issue list above. Huge thanks for that :)
* Let me know if you need additional info.

[image]
[image]
[image]
[image]

PaulWieland, 2023-11-24:
You can see right in the logs that your mqtt broker is sending it commands.

Maybe you sent retained command messages to your broker? Those commands are then sent to ratgdo as soon as it comes online. Clear all retained messages from your broker.

monkitrainer, 2023-11-24:
@PaulWieland Yup, it was a quick fix. Thanks again for your help!

closed 2023-11-24 by monkitrainer