ratgdo/mqtt-ratgdo #12: Periodic status changes when door not opened/closed

lukescott, 2023-11-26:
I have two Security 1.0 doors integrated into HomeAssistant. I'm getting periodic status changes where it will say the door opened, but then immediately closed. Here is some logs from HomeAssistant:

[image]
[image]

During this time I have not opened the garage door, and as you can see, the time in-between status changes are very short.

PaulWieland, 2023-11-26:
A few people have seen this. Not much can be done about the door reporting the wrong status. I could have it ignore short status changes, but then it would add a delay in updating whenever the door state really changes.

closed 2023-11-26 by PaulWieland

lukescott, 2023-11-27:
@PaulWieland Is there anything we can do to look into this a little deeper? This is a fairly big problem. I can't have this reporting the door is open when it's not. If the door actually opens, then I'm bound to ignore it as a false positive.

The only other thing I can think of it the dummy button has some capacitors in it. I'm going to try and disconnect one and monitor it from there. But if this issue persists, I need a way to debug this and get to the root cause.

Snuffy2, 2023-12-12:
I know this is already closed, but I'd agree that it would be helpful if something could be done to filter these out.

[image]

Snuffy2, 2023-12-12:
For now, I've put a Template Cover in HA that is now what Automations and HomeKit are using to reduce false alerts.

```yaml
cover:
  - platform: template
    covers:
      garage_door_bikes:
        device_class: garage
        friendly_name: "Garage Door - Bikes"
        unique_id: garage_door_bikes_safe
        value_template: >-
          {% if is_state('cover.garage_door_bikes_mqtt',states('cover.garage_door_bikes')) %}
          {{ states('cover.garage_door_bikes') }}
          {% elif (is_state('cover.garage_door_bikes','opening') and is_state('cover.garage_door_bikes_mqtt','open') and (as_timestamp(now()) - as_timestamp(states.cover.garage_door_bikes_mqtt.last_changed)) <= 5) %}
          opening
          {% elif (is_state('cover.garage_door_bikes_mqtt','open') and (as_timestamp(now()) - as_timestamp(states.cover.garage_door_bikes_mqtt.last_changed)) > 5) or is_state('cover.garage_door_bikes_mqtt',['opening','closing','closed']) %}
          {{ states('cover.garage_door_bikes_mqtt') }}
          {% else %}
          closed
          {% endif %}
        open_cover:
          - service: cover.open_cover
            target:
              entity_id: cover.garage_door_bikes_mqtt
        close_cover:
          - service: cover.close_cover
            target:
              entity_id: cover.garage_door_bikes_mqtt
        stop_cover:
          - service: cover.stop_cover
            target:
              entity_id: cover.garage_door_bikes_mqtt      
```

Snuffy2, 2023-12-15:
@lukescott thanks! 2023.12.2+ has indeed improved the status of the MQTT Garage Door.