ratgdo/mqtt-ratgdo #13: ratgdo 2.5 /Security 1.0/dumb button MQTT Light problems

cribbstechnologies, 2023-11-28:
Just hooked it up today and the open/close/opening/closing status work great. The obstruction detection works great. Opening/closing works great. If I turn the light on in Home Assistant it comes on but turns off in a few seconds. MQTT output looks like this

```
Topic: gdoratgdo-23664/command/lightQoS: 0
on
2023-11-26 22:16:17:152

Topic: gdoratgdo-23664/status/lightQoS: 0
on
2023-11-26 22:16:18:050

Topic: gdoratgdo-23664/status/lightQoS: 0
off
2023-11-26 22:16:27:776
```

Seems ratgdo doesn't keep track that it was told to turn on and just updates based on whatever the wall button tells it.

I started digging through code and it looks like that's exactly what it's doing.

The status update loop calls this https://github.com/ratgdo/mqtt-ratgdo/blob/2.5/src/ratgdo.cpp#L256 which sets the status based on what it's reading from the serial port.

Then it does the dry contact loop

Then this: https://github.com/ratgdo/mqtt-ratgdo/blob/2.5/src/ratgdo.cpp#L444

The dumb switch doesn't do any keeping of state, does it?

PaulWieland, 2023-11-28:
The wall button doesn't tell ratgdo anything, the door opener does. If you can share the serial log I can have a look at whats gonig on.

cribbstechnologies, 2023-11-28:
> I wonder if some of the troubleshooting we did at this issue might help:
> #8 <https://github.com/ratgdo/mqtt-ratgdo/issues/8>
>
> Some of this sounds similar to the errant behavior of the light on a
> security 1.0 gdo we noted in that ticket.

It's the same wall button as in that issue

jpsutton, 2023-12-15:
I have observed a similar issue, sec+1, same type of dumb button as in the previously mentioned issue. In my case, the light was on, the wall button didn't turn it off. Turning it off from HA worked, but then a few seconds later it would turn on again. It eventually timed out and turned off on it's own.

edrikk, 2023-12-22:
You may want to try the latest build.  It's quite likely that this missing equal sign was the issue.

https://github.com/ratgdo/mqtt-ratgdo/commit/855dc8d5c432894815af2ce8d94bb930f62f5d5c#diff-eb8d72295e2ba240be79284d727e11c809b444e218accf5298b02370e1c554e4R600

The bug was introduced in [560d12b](https://github.com/ratgdo/mqtt-ratgdo/commit/560d12b4ea8ff4fe62ea49e9bdf9c0bdfb59e1ee)

cribbstechnologies, 2023-12-22:
Yep, this fixed it.

> You may want to try the latest build. It's quite likely that this missing equal sign was the issue.
> 
> [855dc8d#diff-eb8d72295e2ba240be79284d727e11c809b444e218accf5298b02370e1c554e4R600](https://github.com/ratgdo/mqtt-ratgdo/commit/855dc8d5c432894815af2ce8d94bb930f62f5d5c#diff-eb8d72295e2ba240be79284d727e11c809b444e218accf5298b02370e1c554e4R600)
> 
> The bug was introduced in [560d12b](https://github.com/ratgdo/mqtt-ratgdo/commit/560d12b4ea8ff4fe62ea49e9bdf9c0bdfb59e1ee)

closed 2023-12-22 by cribbstechnologies