ratgdo/mqtt-ratgdo #15: Possibly incorrect delay for secplus1 in toggleDoor()

p-jean, 2023-12-02:
Found while reading the code in ratgdo.cpp:

```
void toggleDoor(){
    if(controlProtocol == "drycontact"){
        pullLow();
    }else if(controlProtocol == "secplus1"){
        uint8_t delayLen = (lastRX + 275) - millis();
        delay(delayLen);
```

The actual value of delayLen may not be as intended:
1)	Due to the uint8_t truncation, the max delay will be 255. 
        If the value of expression exceeds 255 it will wrap around to a very short delay.
2)	If the expression resolves to a negative value (when lastRx was long ago) then the actual delay will be 0..255 when it probably should be 0 (I assume).

I don’t know enough to predict whether this causes any undesirable outcome.
Curious: 
a) Why does the delay specify 275 ms from the most recent valid RX, rather than 275 ms from the most recent bus activity (including unrecognized RX)? See line 227 in the same file. Should it be called on any RX?
b) Is it correct that this version of RATGDO (secplus1) “hears itself” and so all TX is ultimately RX-ed?

See ratdgo.cpp:714 (commit 45b9263), at the beginning of the toggleDoor() function.

PaulWieland, 2023-12-05:
Thanks for that! Will fix in next release.

PaulWieland, 2023-12-05:
[Corrected](https://github.com/ratgdo/mqtt-ratgdo/commit/346bfbd61744c56e58e9ddabb2d73304195d5c75#diff-eb8d72295e2ba240be79284d727e11c809b444e218accf5298b02370e1c554e4L714)

closed 2023-12-05 by PaulWieland