ratgdo/mqtt-ratgdo #17: RATGDO not controlling opener

timwing, 2023-12-06:
I have had RATGDO's installed, configured and working well for a couple weeks on two identical 2018 Chamberlain B970 MyQ openers. Just today one of them stopped controlling the opener. Both are configured with MQTT & HA. When I send a command to RATGDO from HA I see the RATGDO blue LED flash. I also see sent commands in MQTT. I checked the wire connections and they are sound. I have rebooted the RATGDO and removed power from the opener and it's backup battery.

If I watch the RATGDO door controls in HA and press the garage door opener button on the wall, I see the HA RATGDO status update so it is receiving state updates but the opener just wont respond to commands sent by RATGDO.

timwing, 2023-12-09:
As of today it started working again all on it's own. What would cause this?

PaulWieland, 2023-12-09:
Try upgrading to v2.53 (assuming you have a v2.5 board)

coreywilner, 2023-12-20:
I found the BIN file for 2.55 and installed it with the OTA feature from the ratgdo main page.  I too had the problem with the unit that status was showing up ok, but control was not happening.  With 2.55, the unit is now working again for controlling the open/close of the ratgdo.

timwing, 2023-12-20:
I tried the "ratgdo_esp8266_hV25_sV2.55.bin" but I get "Update error: ERROR[10]: Magic byte is wrong, not 0xE9" so I am now using 2.53.

msl2424, 2023-12-21:
I have the same issue. 

On 12/9, I installed Ratgdo v2.5 on a Chamberlain MyQ garage door opener (Model B4505) with a yellow learn button and Security+ 2.0 flashed with MQTT. Ratgdo was auto-discovered by MQTT in Home Assistant, and I could open and close the garage door and turn the built-in light on and off from Home Assistant. I could also open and close the garage door using the Chamberlain wall switch inside my garage, the wireless numeric keypad outside my garage, the garage remote, the MyQ iOS app, and the programmed button in my car – everything worked great.

However, sometime in the past couple days, I could no longer open and close the garage door or turn the built-in light on and off using Home Assistant or the wireless numeric keypad outside my garage. Everything else (wall switch inside my garage, garage remote, MyQ app, car button) can still open and close the garage door, and Home Assistant receives real-time status of the garage and obstruction sensor, and my automations notifying me when the garage door opened or closed still work.

To troubleshoot, I re-did the wiring same as before, but I still can’t regain the ability to open/close and toggle the light in Home Assistant (and I don’t know why my numeric keypad outside the garage stopped working as well).

Attached are photos of my setup. Any ideas on what to do to get this working so I can control via Home Assistant and MQTT?

[image]

[image]

[image]

[image]

[image]

[image]

PaulWieland, 2023-12-21:
Make sure you are running firmware v2.55

msl2424, 2023-12-21:
Where can I download the v2.55 .bin file for an OTA update? I don't see it here: https://github.com/PaulWieland/ratgdo/releases

msl2424, 2023-12-22:
I was able to get v2.55 from someone in the HA community forum, but updating to v2.55 did not solve the issue. I still cannot  open/close the garage door from HA, but HA does show real-time status and the physical wall mount garage button works.

timwing, 2023-12-22:
@msl2424 Does operating the door a couple times with the remote help? I cant be sure but mine started working again without any firmware updating.

coreywilner, 2023-12-22:
Even with 2.55, the ratgdo has reverted to simply reporting status and not responding to control actions. 
Do you have any new fixes for this defect?

msl2424, 2023-12-22:
> @msl2424 Does operating the door a couple times with the remote help? I cant be sure but mine started working again without any firmware updating.

I opened and closed it three times in a row with the remote. The remote worked but I still can't open and close using Home Assistant (but it reports the correct door status).

PaulWieland, 2023-12-22:
> Even with 2.55, the ratgdo has reverted to simply reporting status and not responding to control actions. Do you have any new fixes for this defect?

You need to post a serial log.

coreywilner, 2023-12-22:
> > Even with 2.55, the ratgdo has reverted to simply reporting status and not responding to control actions. Do you have any new fixes for this defect?
> 
> You need to post a serial log.

Where are there instructions on how to grab a serial log?  Can this be done remote or do I have to connect to the ratgdo directly?

coreywilner, 2023-12-22:
> Where are there instructions on how to grab a serial log? Can this be done remote or do I have to connect to the ratgdo directly?

Interesting...a full reboot of Home Assistant and the ratgdo (2.55 MQTT version) is working again. Not sure the reason for this as all my other MQTT devices are not having issues with subscriptions.  MQTT explorer shows that a command is published (for open or close) but ratgdo is not apparently acting on it.  And it is publishing status which I can see changing in MQTT Explorer, and it does so with commands from MyQ app or keypad or HomeLink.

sejgit, 2023-12-22:
coming and going as well.  with a 3rd reboot & 5 min wait it came back.
went along with the light status on mqtt showing unknown or [name of unit]/status/light when I turned it on or off manually.

jmedemar, 2023-12-27:
Having the same issue here. Just got my revision 2.5 boards and loaded 2.56 firmware. Can see the status of the door, light, and obstruction sensors, but cannot send a command to the door. Are we supposed to be using a topic of "home/garage/ratgdo-65546/**status**/door" or "home/garage/ratgdo-65546/**command**/door" for sending the "open" or "close" commands via mqtt? The light responds using the topic of "status" but not "command". Cannot get the door to respond by using either topic.

PaulWieland, 2024-01-02:
Anyone having this issue needs to post a serial log so we can see what is going on.

- Connect to the board with a USB cable
- go to logs and console on the esp web tools page
- click reset device to start a fresh log
- reproduce your issue
- attach the log here

treylok, 2024-01-08:
Same problem here. Status reads correctly, but commands will not open or close the door or control the light.  Attached is a log.

[esp-web-tools-logs.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13860566/esp-web-tools-logs.txt)

PaulWieland, 2024-01-08:
> Same problem here. Status reads correctly, but commands will not open or close the door or control the light. Attached is a log.
> 
> [esp-web-tools-logs.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13860566/esp-web-tools-logs.txt)

Did it work before and then stop working? Just to rule out a client ID issue, re-install with full erase so it generates a new client ID. You can also install the ESPHome version and test the controls or assign new client ID from the Web UI.

treylok, 2024-01-08:
> Did it work before and then stop working? Just to rule out a client ID issue, re-install with full erase so it generates a new client ID. You can also install the ESPHome version and test the controls or assign new client ID from the Web UI.

It did work before.  It worked with original firmware shipped with unit, then stopped working.  After flashing to 2.55 it was working again, then stopped.  Flashing to 2.57 did not fix.

I will try a full erase, then ESPHome version.

Thanks.

PaulWieland, 2024-01-08:
Just to be clear, no ratgdo hardware ships with firmware pre-installed.

treylok, 2024-01-08:
> Just to be clear, no ratgdo hardware ships with firmware pre-installed.

Sorry, yes, it was not pre-installed, just forgot initial procedure.  I am not sure which version it was though.

I ended up just flashing to ESPHome and it seems to be working now.

brandonleon, 2024-01-11:
I am experiencing a similar issue. It worked for the first few days, but now it is correctly reporting the status to MQTT, yet it ignores all commands. I can see the commands hitting MQTT and the state changes for the light in HA, but nothing actually happens.

PaulWieland, 2024-01-11:
If you are using Home Assistant and you have a Security + 2.0 opener, you really should be using the ESP Home firmware...
That being said, try re-installing with erase to see if a new client ID fixes the problem. Again, you need to post a serial log otherwise we're just guessing at what is going on.

brandonleon, 2024-01-11:
> If you are using Home Assistant and you have a Security + 2.0 opener, you really should be using the ESP Home firmware... That being said, try re-installing with erase to see if a new client ID fixes the problem.

Thanks, I already have MQTT devices in Home Assistant, which is why I went that route. I went ahead and installed the ESPHome firmware, and everything is working like new again. Thanks for an amazing little device.

john-kapp, 2024-01-15:
@PaulWieland I believe I have this same problem.  My Security+ 2.0 opener worked well for about a week after the initial installation of v2.57, but then I stopped being able to control the door via HA although HA still reported status changes (door open/close, motion, etc.).  Simply restarting RatGDO had no effect on the issue.   Per your message above, I reinstalled v2.57 with an erase and HA was then able to control the GDO once again.  I have attached four log files.  The first two are before the erase and the second two are after the erase.  One using HA to try to control the opener and one using the wall button.  I hope this are helpful in debugging.  Thanks for all your work on this.

[file1-ha-open-command-nowork.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13942431/file1-ha-open-command-nowork.txt)
[file2-wallbutton-open-works.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13942434/file2-wallbutton-open-works.txt)
[file3-post-erase-wallbutton-open-works.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13942435/file3-post-erase-wallbutton-open-works.txt)
[file4-post-erase-ha-open-command-works.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13942438/file4-post-erase-ha-open-command-works.txt)

sejgit, 2024-03-19:
Same issue.  We are all using the MQTT version.  Works for a bit then stops controlling.  Works again with a refresh but not just a power cycle.

geozahar, 2024-03-20:
Same issue.  Using Security + 2.0 opener with MQTT firmware.  Have to erase/flash (using latest) periodically to regain control.
If you still need more logs.  Let us know.

graydavidw, 2024-03-20:
I now think that I’ve had this issue as well- originally I thought it was part of a crash I had with my system (Homeseer plugin for MQTT)  but then had it again this week.

Ishfal, 2024-03-22:
+1 to this issue

hart-2-hart, 2024-03-25:
I read where others are conjecturing that power issues like failure or brown-outs may affect the electronics and result in this issue. Both of my ratgdo had this issue at the same time several weeks after install.  My electric service is underground and to fix a sinking conduit that was pulling free from meter, the utility company cut power  for a repair.  Not saying it is connected but just that both doors had issue at same time, and I'd had a full outage and restore of power.

If this issue is not found, I may locate access to the UPS battery in one of the openers and draw power from it instead of AC wall wort.  I can then test if both get issue or only the one on AC.

mlamoure, 2024-04-10:
Piling on with my experience:

- Ratgdo stopped being able to control my door sometime in mid March 2024, several months after initial configuring and wiring.
- All door states were reporting properly from Ratgdo to MQTT Server.  Only controlling the door stopped working.
- I could see MQTT getting the commands from my various MQTT publishers that were sending commands.
- Climbing up to where the device was mounted, I could see the Ratgdo light turning on when I send a door command via MQTT, but nothing was happening.
- I took pictures of my wiring, disconnected, re-flashed, and re-wired.  Controlling the door resumed
- Three days later, the same behavior returned.  Unlike some here, powering down and back up does not resume functionality.

If there's anything I can do to help diagnose, please let me know.

hart-2-hart, 2024-04-12:
> * esp web tools

Second time with issue.  I have  log for one of two ratgdo where it stopped responding but continued to update status.  First time it happened both had the issue but unsure exactly when it occurred.  How do you want to receive log.  Not comfortable just posting here unless you confirm nothing sensitive is included.

gmauleon, 2024-07-26:
Poking here as well with the same problem.
Using Security + 2.0 opener with MQTT firmware 2.57, worked a bit, then all commands stopped working, all status still works.

I understand ESPHome seems to work, will probably try but still interested to use MQTT as all my others devices goes trough MQTT.

Thanks in advance!

PaulWieland, 2024-07-26:
Reflash with full erase to get a new randomly generated Client ID. Alternatively you can make your own yaml for ratgdo esphome that enables mqtt.

mabordas, 2024-07-26:
Reflashing only works for a few hours and then the problem comes back, so that's not a great solution. Are we saying the esphome firmware with mqtt enabled is going to work better? If that's true, why can't we get those changes into this project?

ISMiller1, 2024-10-01:
Has this issue been solved yet? I am running V2.57 with MQTT and having same issue. Home Assistant and RATGDO recognizes the real-time sensor status (obstruction, light, door), but none of the commands work through Home Assistant (door open/close, or light on/off). I originally installed back in April 2024 and everything was working great until just a few days ago...

PaulWieland, 2024-10-01:
I _finally_ have an example of this problem in front of me so I can trouble shoot and fix it, and plan to do so in the next few days.

closed 2024-10-04 by PaulWieland

mabordas, 2024-10-08:
@PaulWieland  Apologies if this is a silly question, but how do we install 2.59? I don't see it in the releases: https://github.com/ratgdo/mqtt-ratgdo/releases

BobbyBleacher, 2024-10-26:
Just installed, does not fix the issue.

fmeili1, 2024-11-08:
> Has this issue been solved yet? I am running V2.57 with MQTT and having same issue. Home Assistant and RATGDO recognizes the real-time sensor status (obstruction, light, door), but none of the [...]

I have the exact same problem on just one of my 2 garage door openers (1xRVGarage 1xCarGarage). Both worked since about 6 month with the v2.57 (hardware v2.53i) without any issues. Since a couple of days, the RV opener behaves like you've described (status works, but no controlling) the car garage opener is still working without issues.

I've updated both openers to v2.59 and still the same issue with the v2.59 on the RV garage opener - car garage opener works correct also with the v2.59.

- RV garage opener is a "LiftMaster 8587WL" (not working)
- Car garage opener is a "LiftMaster 8355W" (working)

ISMiller1, 2024-11-08:
I installed 2.59 on October 5, and it's been working great since. Thank you.

lilyball, 2024-11-13:
I just started having this issue. The ratgdo has been working fine for a while, but it recently stopped. I was on the 2.57 firmware, I just updated to 2.59, still nothing. I haven't physically inspected the wiring since that's super awkward but it was installed very securely to begin with, and it is reporting the current garage door state just fine, it just can't control it, not even toggling the light.

I see suggestions here to reflash with erase. This is super awkward to do, getting the thing installed in the first place was a PITA and taking it out again in order to plug into my computer before reinstalling would be even harder. Is there no way to get a new client ID with the web config?

Similarly I see a recommendation to switch to the ESPHome firmware. Besides the above problem, I've never used ESPHome but I have used MQTT, which is why I'm using the MQTT firmware. I'm also confused as to why the ESPHome firmware would be sending any different control codes than the MQTT firmware; surely the interface for communicating with HA is orthogonal to the protocol for communicating with the garage door lifter?

fmeili1, 2024-11-16:
> Looks like I'm not the only one who still have this problem, even after updating to 2.59. Any ideas how to fix this?

I would like to help to analyze the problem, but I don't know?

- Is it possible to enable advanced logging (I don't see such an option in the MQTT WebGUI)?
- Would it help to temporarily install the usual firmware (not the MQTT one) to do further analyzing of the problem?

Thanks in advance

fmeili1, 2024-11-21:
I've found something which may be of interest about this error.

Usually I use the command line mqtt client to watch/debug the messages with grep to specific prefix topics.
Now I've installed (for other reasons) the graphical MQTT Explorer and found that I have 3 main topics for my 2 ratgdo garage door openers.

The still working 2 car garage door opener sends out two different topic prefixes, even if I've only configured the prefix "ratgdo2car":
- /ratgdo2car/status/...
- /adminratgdo2car/status/...
Both have the same attributes like "door", "light", etc.

But the not longer working RV garage door opener sends out only one topic:
- /ratgdorv/status/...

I don't know where the prefix "/adminratgdo2car" is coming from but this is a visible difference compared to the not longer working ratgdo for my RV garage door.

Maybe this could point to the problem why one opener is working but the other not?!

fmeili1, 2024-12-11:
Today I've noticed that the same problem occur now on my 2-car garage door opener also. Now both RATGDO are not longer working to control the doors. Even worse, even the state of the doors is not longer provided (this worked for my RV garage door for a while, even it was not possible to control it since initial problem occured).

Is it possible that one of the myQ app software updates on my mobile phone changed the garage door openers (firmware) in a way that they are not longer compatible with RATGDO - just a question.

fmeili1, 2024-12-13:
@PaulWieland 

I've just noticed that I've put my comments/notes/problems on this issue which is already closed. I found no way to re-open this issue by myself.

Because now both of my ratgdo's are not longer working (after many month without any problems).

Please can you reopen this issue?
Thanks!