ratgdo/mqtt-ratgdo #19: Suggestion: make wall-panel emulation into a configurable feature

p-jean, 2023-12-07:
Wall panel emulation on secplus1 should be configurable rather than automatically entered on the grounds:
1) Most people will not use it (a guess)
2) It is something that can be known ahead of time
3) Entering wall panel emulation accidentally will cause bus collisions
4) Wall panel emulation mode is a dead-end until reset

PaulWieland, 2023-12-07:
Wall panel emulation is required if you do not have a wall panel which queries for door status, so it absolutely is not optional. 

Emulation mode is only entered when a smart wall panel isn’t detected, so if you are having collisions it means the detection didn’t work for some reason. 

What wall panel do you have and what commands is it sending to the gdo?

p-jean, 2023-12-08:
My suggestion is to make it into a configurable item, like the protocol, so that those who need it can have it.

closed 2023-12-09 by PaulWieland