ratgdo/mqtt-ratgdo #23: 2.5i board on 2.53, has never actually controlled a thing; doesn’t work!

pcm1ke, 2023-12-10:
Good evening Paul,

I flashed my new ratgdo with your MQTT firmware, but I was never able to get the homebridge plugin to actually work with it, despite everything being set properly.

The plugin was able to detect when there was motion, it was able to detect when the door was opened/closed, and able to detect if the light was turned off/on (both of these via the wall switches), but when the accessories exposed in homebridge were toggled the ratgdo didn’t actually cause the gdo to do a thing. It’s worth noting that the blue LED would light up each time, so presumably the MQTT messages were making it to the ratgdo, but nothing would ever happened. I also loaded my own MQTT client and tried publishing the same messages manually and again the ratgdo blue LED would light up, but nothing would ever happen.

This is a little bit of a bummer as homebridge doesn’t have an esphome plugin like home assistant does and I didn’t really want to run home assistant just to use esphome to control a single device, but as of right now it looks like that’s the only way I’m going to be able to control the ratgdo?

I’ve tried erasing and reflashing the mqtt firmware a few times, not just a couple and still no dice.

It’s worth noting, I doubt the issue has anything to do with my wiring as esphome under home assistant functions perfectly.

Thank you,
Michael

PaulWieland, 2023-12-10:
What does the ratgdo serial log show?

meach213, 2023-12-11:
Paul,
  I am having same issues. I have the logs here. If you have any questions or need me to test anything else please let me know!

First few attempts were from Homekit. Last couple were from the wall switch.

QUEUE MSG ARRIVED [ratgdo-dual/command/light] 
off
MQTT: turn the light off
rolling code for 0 32|light : 5501005908200400041201252DB65B2D761B0E
rolling write successful
5501005908200400041201252DB65B2D761B0E | LIGHT:0
Light state off
QUEUE MSG SENT [ratgdo-dual/status/light] 
off
QUEUE MSG ARRIVED [ratgdo-dual/command/light] 
on
MQTT: turn the light on
rolling code for 0 33|light : 5501005825964D24965B24161B7FBEDF0DD7EA
rolling write successful
5501005825964D24965B24161B7FBEDF0DD7EA | LIGHT:1
Light state on
QUEUE MSG SENT [ratgdo-dual/status/light] 
on
QUEUE MSG ARRIVED [ratgdo-dual/command/light] 
off
MQTT: turn the light off
rolling code for 0 34|light : 55010002092CBACB24925986192CB2DB8DA647
rolling write successful
55010002092CBACB24925986192CB2DB8DA647 | LIGHT:0
Light state off
QUEUE MSG SENT [ratgdo-dual/status/light] 
off
QUEUE MSG ARRIVED [ratgdo-dual/command/door] 
open
MQTT: open the door
rolling code for 0 35|door1 : 5501000102490C104924956804904120E0412C
rolling code for 0 35|door2 : 5501000102490C104924946804904120E0512C
rolling write successful
5501000102490C104924956804904120E0412C
5501000102490C104924946804904120E0512C
QUEUE MSG ARRIVED [ratgdo-dual/command/door] 
close
MQTT: close the door
The door is already closed
QUEUE MSG ARRIVED [ratgdo-dual/command/door] 
open
MQTT: open the door
rolling code for 0 36|door1 : 5501009036DB67B65B2DB55525B2596D72CB3C
rolling code for 0 36|door2 : 5501009036DB67B65B2DB75525B2596D72EB3C
rolling write successful
5501009036DB67B65B2DB55525B2596D72CB3C
5501009036DB67B65B2DB75525B2596D72EB3C
QUEUE MSG ARRIVED [ratgdo-dual/command/door] 
close
MQTT: close the door
The door is already closed
550100840A1D91EFBFFF7F682DF337ACE0512E | LIGHT:1
Light state on
QUEUE MSG SENT [ratgdo-dual/status/light] 
on
550100822E8EFC596492595912C6ED11C43215 | LIGHT:0
Light state off
QUEUE MSG SENT [ratgdo-dual/status/light] 
off
550100513C8F78934D3493482D70362C3A5D28 | LIGHT:1
Light state on
QUEUE MSG SENT [ratgdo-dual/status/light] 
on
55010050111931B7DF7DB02925C1ECA1C08221
55010050111931B7DF7DB42925C1ECA1C0C221
55010045398EBBEAACDBE74618BD74BBC5A60F | STATUS: door:4 light:1 lock:0 obs:1
Door state opening
QUEUE MSG SENT [ratgdo-dual/status/door] 
opening
55010045398EB9EABEDBEF4618BD74BBC1374B
55010045398EB9EABEDBEF4618BD74BBC1374B
55010045398EB9EABEDBEF4618BD74BBC1374B
5501004439CF9F78F7F77B2828D86075D18B22
5501001228608A4A205279163958A25305F7F8 | STATUS: door:1 light:1 lock:0 obs:1
Door state open
QUEUE MSG SENT [ratgdo-dual/status/door] 
open
550100A60DAC1A934D3693980DE17FAC72112E | LIGHT:0
Light state off
QUEUE MSG SENT [ratgdo-dual/status/light] 
off
550100A53A577725965B2C8919E35B0AE0140E | LIGHT:1
Light state on
QUEUE MSG SENT [ratgdo-dual/status/light] 
on
55010014141F69FFFFBEFC682D717EACE0412C
55010011138D3C914966970830F450AEB08D21
55010014141F69FFFFBEFD682D717EACE0512C
550100680638690789412C661E25148B1FB7B8 | STATUS: door:5 light:1 lock:0 obs:1
Door state closing
QUEUE MSG SENT [ratgdo-dual/status/door] 
closing
5501006806386B07804124661E25148B1B377B
5501006806386B07804124661E25148B1B377B
5501006806386B07804124661E25148B1B377B
550100661D5796F876B7F35827070DC55F7B30 | STATUS: door:2 light:1 lock:0 obs:1
Door state closed
QUEUE MSG SENT [ratgdo-dual/status/door] 
closed

meach213, 2023-12-11:
Another issue that i dont think is your problem is the Light status seems to be the only one that doesnt track properly with the Homebridge plugin. I see the update in the logs but the homebridge plugin doesnt update correctly, The door status does though.

pcm1ke, 2023-12-11:
FYI, my GDO is a Chamberlain B970

meach213, 2023-12-11:
Mine is a liftmaster 81650MC

tbutkiewicz, 2023-12-14:
I have the same issue.  Ratgdo is able to track status of all the sensors and things from GDO, but is unable to change any of them.  

- If I initiate a command from ratgdo > GDO = no change, hangs.  
- Initiating same command from GDO/panel = ratgdo tracks status change correctly.  

Changes only flow from GDO back to ratgdo. Commands _originating_ from ratgdo have no change/impact and hang.

Hardware:
- [CRAFTSMAN 1-1/4 Hps myQ Smart Belt Drive Garage Door Opener Model #CMXEOCG982](https://www.lowes.com/pd/Chamberlain-1-25-HP-myQ-Belt-Drive-Garage-Door-Opener/5001544439) re-branded Liftmaster
- And [LiftMaster 880LM Smart Control Panel Security + 2.0 for Yellow Learn Button](https://www.amazon.com/gp/product/B07Q2S57FV/)

pcm1ke, 2023-12-14:
Tom,

Have you tried switching firmware to esphome and leaving everything else
the same and see if it works? I know this isn’t the ideal answer, because
you probably want the simplicity that is mqtt, but I’m curious if your
results.

Michael A. Nunes
/p at pcmike dot net

thewazz1, 2023-12-14:
I have the same issue, but it comes and goes. When I first flashed it and
hooked it up it worked. Stopped the next day. A few days later it worked
for a day, and is now broken again. I see the state changes, but commands
don’t do anything other than flash the blue light on the ratgdo.

tbutkiewicz, 2023-12-15:
> Tom, Have you tried switching firmware to esphome and leaving everything else the same and see if it works? I know this isn’t the ideal answer, because you probably want the simplicity that is mqtt, but I’m curious if your results.

Thanks for the suggestion Michael - just flashed it with the esp version and.... **it works**!  No wiring changes or anything.

I did grab some console output of failed triggers for light on/off and failed opening of garage door when still on the mqtt firmware:
```
QUEUE MSG ARRIVED [/home/garage/ratgdo/command/light] 
off
MQTT: turn the light off
rolling code for 0 461|light : 55010019001045040048046529B69B69C4CA
rolling write successful
55010019001045040048046529B69B69C4CA | LIGHT:3
Light state /home/garage/ratgdo/status/light
QUEUE MSG SENT [/home/garage/ratgdo/status/light] 
/home/garage/ratgdo/status/light

QUEUE MSG ARRIVED [/home/garage/ratgdo/command/light] 
off
MQTT: turn the light off
rolling code for 0 462|light : 550100842FBFFDEFBFFF7F141FEFF7FF9F9E
rolling write successful
550100842FBFFDEFBFFF7F141FEFF7FF9F9E | LIGHT:2
Light state unknown
QUEUE MSG SENT [/home/garage/ratgdo/status/light] 
unknown

QUEUE MSG ARRIVED [/home/garage/ratgdo/command/door] 
open
MQTT: open the door
rolling code for 0 463|door1 : 550100820B2C905964925C053DF6FB6F16DB
rolling code for 0 463|door2 : 550100820B2C9059649258053DF6FB6F169B
rolling write successful
550100820B2C905964925C053DF6FB6F16DB
550100820B2C9059649258053DF6FB6F169B
```

PaulWieland, 2023-12-15:
> rolling code for 0 463|door1

for some reason your client ID is zero, which it shouldn’t be. What version of the firmware was that from?

pcm1ke, 2023-12-15:
Paul, in all cases after flashing the mqtt firmware and visiting the ratgdo device page, I had changed the “semi-random” default name to simply “ratgdo” no numbers or letters, could this lead to ending up with a client id of 0 causing this to happen?

PaulWieland, 2023-12-15:
No. The device name is not related to the client id. Until I figure out what’s going wrong with the client id initialization you can try and do a reinstall with a full erase.

tbutkiewicz, 2023-12-15:
> > rolling code for 0 463|door1
> 
> for some reason your client ID is zero, which it shouldn’t be. What version of the firmware was that from?

My console output was from the latest version available on the flash web ui - `ratgdo v2.54 for hardware v2.50`. 

I did reflash multiple times during troubleshooting, all resulted in the same outcome when on the mqtt version.

atwoodjw, 2023-12-18:
I'm seeing the same issue (ratgdo v2.54 for hardware v2.50). I too seem to have client ID = 0. I've re-flashed 3x now.

```
QUEUE MSG ARRIVED [right-garage-door/command/door] 
open
MQTT: open the door
rolling code for 0 6|door1 : 5501008826934B34D24D218034DB6DB63B6DBE
rolling code for 0 6|door2 : 5501008826934B34D24D258034DB6DB63B7DBE
rolling write successful
5501008826934B34D24D218034DB6DB63B6DBE
5501008826934B34D24D258034DB6DB63B7DBE
QUEUE MSG ARRIVED [right-garage-door/command/door] 
close
MQTT: close the door
rolling code for 0 7|door1 : 550100860B2CB4CB2DB64E61165924963B249E
rolling code for 0 7|door2 : 550100860B2CB4CB2DB64A61165924963B349E
rolling write successful
550100860B2CB4CB2DB64E61165924963B249E
550100860B2CB4CB2DB64A61165924963B349E
QUEUE MSG ARRIVED [right-garage-door/command/door] 
open
MQTT: open the door
rolling code for 0 8|door1 : 5501001116C921924964951032DB2CB6BB2DA6
rolling code for 0 8|door2 : 5501001116C921924964941032DB2CB6BB0DA6
rolling write successful
5501001116C921924964951032DB2CB6BB2DA6
5501001116C921924964941032DB2CB6BB0DA6
QUEUE MSG ARRIVED [right-garage-door/command/door] 
close
MQTT: close the door
rolling code for 0 9|door1 : 5501001036CB68B6DB2DB50110410090E924B0
rolling code for 0 9|door2 : 5501001036CB68B6DB2DB40110410090E964B0
rolling write successful
5501001036CB68B6DB2DB50110410090E924B0
5501001036CB68B6DB2DB40110410090E964B0
```

a1224, 2023-12-19:
> I'm seeing the same issue (ratgdo v2.54 for hardware v2.50). I too seem to have client ID = 0. I've re-flashed 3x now.

+1, tried re-flashing with erase multiple times, all ending up with client_id=0

FW: 2.54 
Board: 2.5
GDO: Liftmaster 8550

PaulWieland, 2023-12-19:
Ok for those getting a client id of zero, please flash with full erase and capture the serial log from initial bootup.

PaulWieland, 2023-12-19:
Scratch that, I was finally able to figure out whats wrong. Please stand by for v2.55

PaulWieland, 2023-12-19:
Please install `v2.55`. A full erase shouldn't be necessary, it detects invalid client IDs upon startup and then initializes a new one if necessary.

After the setup, look for `Initializing new client ID`:

```
Using security+ 2.0
Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.55
IMPROVhttp://10.0.1.113S
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: test/GlassDoor2/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [test/GlassDoor2/command/#] 

...

QUEUE MSG SENT [test/GlassDoor2/status/availability] 
online
idCode doesn't exist. creating...
idCode write successful
exisiting client ID: 0
Initializing new client ID: 
idCode write successful
1B9539
rolling doesn't exist. creating...
rolling write successful
Syncing rolling code counter after reboot...
rolling code for 1B9539 0|reboot1 : 5501000036DB2DB7DB6DB6001749FEB3E96FBB
rolling code for 1B9539 1|reboot2 : 550100AA2CB2596F965961A0144C48B13BADB6
```

meach213, 2023-12-19:
You the man Paul! works like a charm now!

a1224, 2023-12-19:
Awesome, thanks for the quick turnaround :)

atwoodjw, 2023-12-19:
Confirmed. Works great. Thanks!

closed 2023-12-19 by PaulWieland

wisteso, 2023-12-19:
I was looking at the commits for 2.55 and while they're nice changes, I don't how it can fix the issue for people who had been already erasing the flash memory upon update.  

I suppose most people having this issue could have been forgetting to checkmark 'erase' though and thought flashing automatically erased.  PCMike hasn't yet flashed to 2.55 when I talked to him, so I could see the issue maybe remaining.

sejgit, 2023-12-22:
> I have the same issue, but it comes and goes. When I first flashed it and hooked it up it worked. Stopped the next day. A few days later it worked for a day, and is now broken again. I see the state changes, but commands don’t do anything other than flash the blue light on the ratgdo.

I have the same coming & going issue.  didn't work in 2.51 and started working in 2.55 but then stopped being able to control a day later.  Still not working with 2.56

john-kapp, 2023-12-23:
I believe I'm seeing the same issue with my Security+ 2.0 opener.  I first started at firmware v2.54 and had the ClientID=0 issue.  I then upgraded to v2.55 including an erase of the RatGDO.  Door control started working immediately.  After a couple of days, I was still getting status messages, but I lost the ability to control the door via the RatGDO.  I then upgraded to v2.56 via the web UI (without an erase) which didn't fix the issue.  I was still getting status message but no control.  

I now just connected my laptop to the RatGDO can captured logs when it wasn't working.  I then reflashed v2.56 with an erase - and now things are working again.  I captured logs with it working in case helpful.  We will see if it stops working again after a couple of days.

Serial logs attached.
[Before erase - not working.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13759894/Before.erase.-.not.working.txt)
[After erase - working2.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13759896/After.erase.-.working2.txt)