ratgdo/mqtt-ratgdo #25: HA not detecting Ratgdo MQTT

mrwkbrdr, 2023-12-10:
For some reason my HA is not detecting my ratgdo. My HA is reporting other MQTT devices? Yes the auto discovery is selected in MQTT HA.
My MQTT broker can see it?
Any suggestions?

HA:
Core
2023.11.3
Supervisor
2023.11.6
Operating System
11.1
Frontend
202310.2

MQTT broker:

shellyplugus-
shellyplugus-
frigate
homeassistant
cover
light
binary_sensor
MAIN DOORGARAGE DOOR
status
availability = online
obstruction = clear
motion = detected
door = closed
light = on
lock = unlocked
$SYS
shellies

dbrewer-wdc, 2023-12-10:
The autodiscovery stuff should be under homeassistant....for instance, homeassistant/cover/<your device>/config and config would be equal to a string of JSON structure. Here is an example of mine from MQTT Explorer; mine has a few extra devices, so ignore those. You can get MQTT Explorer by just searching for it on the internet, or even in Github. Makes troubleshooting a bit simpler.
[image]

mrwkbrdr, 2023-12-10:
I can see the 3 MQTT configs under home assistant in my MQTT broker, but I have no entities showing up in HA UI?

[image]

mrwkbrdr, 2023-12-11:
Well I re read the info, and found an invalid character that HA refused. All good now.

closed 2023-12-11 by mrwkbrdr

dbrewer-wdc, 2023-12-11:
@mrwkbrdr what was the invalid character?

mrwkbrdr, 2023-12-12:
You cannot have any spaces or special characters.