ratgdo/mqtt-ratgdo #27: Security 1.0 - Opening automatically after power outage

PK7622, 2023-12-12:
I got home today at the garage door was wide open.

There had been a power outage and when the power came back on the door must have opened. In Home Assistant the logs had a constant loop of:

Is closing
5:24:56 PM - 4 hours ago
Is opening
5:24:54 PM - 4 hours ago
Was opened
5:24:54 PM - 4 hours ago
Is closing
5:24:50 PM - 4 hours ago

I closed the door with the wall button only for it to open by itself a few minutes later. Has anyone seen anything like this before? I didn't have any issues before this.

lukescott, 2023-12-15:
What wall panel do you have? I have read reviews of the 889LM doing this on it's own.

PaulWieland, 2023-12-15:
Capture the serial log output from ratgdo. You probably sent a retained door command message.

p-jean, 2023-12-16:
I have the same issue. I raised bug #18 
https://github.com/ratgdo/mqtt-ratgdo/issues/18

p-jean, 2023-12-16:
Seen again with version 2.54

closed 2023-12-17 by PaulWieland

gullygossner, 2023-12-30:
Same scenario on my end, Model RJO20C. It occurred with firmware 2.51 and 2.52. I've had to unplug the ratgdo until a solution (hopefully) is found.