ratgdo/mqtt-ratgdo #28: Sec+1 hangs/crashes every time i try to open/close the door

JeffSteinbok, 2023-12-12:
Just received my unit today, flashed it, and having all sorts of weird issues.
I initially put ESPHome on it, but realized I have a 1.0 opener so flashed back to the MQTT version (2.54)

This is a chaimberlain liftmaster with the Purple button.

Now I’m finding that opening & closing the door via HomeAssistant causes the device to become unavailable and eventually timeout.  The blue LED on the board stays on solid until such time as it timeouts and reboots.

Door Stop, and Light/Obstruction all work fine.

Is there a good way to debug what's going on here?

JeffSteinbok, 2023-12-12:
Here is the serial log; note that this is presently NOT connected to the door, just my PC.  Was doing the same thing when connected though.  Note there is a long hang between trying to open the door and the apparent reboot:

config]
{
  "~": "ratgdo-78904",
  "name": "Obstruction",
  "unique_id": "ratgdo-78904_08:3A:8D:F9:ED:F9_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "ratgdo-78904",
    "identifiers": "ratgdo-78904_xx",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.54",
    "configuration_url": "http://192.168.1.180/"
  }
}
QUEUE MSG SENT [ratgdo-78904/status/availability]
online
QUEUE MSG ARRIVED [ratgdo-78904/command/light]
off
MQTT: turn the light off
QUEUE MSG ARRIVED [ratgdo-78904/command/light]
on
MQTT: turn the light on
QUEUE MSG ARRIVED [ratgdo-78904/command/door]
stop
MQTT: stop the door
The door is not moving.QUEUE MSG ARRIVED [ratgdo-78904/command/door]
open
MQTT: open the door
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params:
willTopic: ratgdo-78904/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [ratgdo-78904/command/#] 
QUEUE MSG SENT [homeassistant/cover/ratgdo-78904/config] 
{
  "~": "ratgdo-78904",
  "name": "Door",
  "unique_id": "ratgdo-78904_08:3A:8D:F9:ED:F9",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "ratgdo-78904",
    "identifiers": "ratgdo-78904_08:3A:8D:F9:ED:F9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.54",
    "configuration_url": "http://192.168.1.180/"
  }
}
QUEUE MSG SENT [homeassistant/light/ratgdo-78904/config]
{
  "~": "ratgdo-78904",
  "name": "Light",
  "unique_id": "ratgdo-78904_08:3A:8D:F9:ED:F9_light",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/light",
  "payload_on": "on",
  "payload_off": "off",
  "state_topic": "~/status/light",
  "device": {
    "name": "ratgdo-78904",
    "identifiers": "ratgdo-78904_08:3A:8D:F9:ED:F9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.54",
    "configuration_url": "http://192.168.1.180/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/ratgdo-78904/config] 
{
  "~": "ratgdo-78904",
  "name": "Obstruction",
  "unique_id": "ratgdo-78904_08:3A:8D:F9:ED:F9_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "ratgdo-78904",
    "identifiers": "ratgdo-78904_08:3A:8D:F9:ED:F9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.54",
    "configuration_url": "http://192.168.1.180/"
  }
}
QUEUE MSG SENT [ratgdo-78904/status/availability]
online

JeffSteinbok, 2023-12-12:
[image]

This is the wall unit I have. Starting to suspect it could be due to that.

lukescott, 2023-12-15:
Here's a potential debugging step. I would try disconnecting the wall buttons, and then reboot the Ratgdo. This should put the Ratgdo in emulation mode. See if if you can control with HA without it rebooting.

JeffSteinbok, 2023-12-31:
So, I think something else may be up. 
Can someone please tell me what is supposed to happen if I have it not wired up at all; just on my desktop, and I hit the open or close buttons in HomeAssistant?  

Whenever I hit Stop or Light, the LED blinks and all is well.
If I press OPEN or CLOSE, the LED stays on, and the device goes unavailable for a period of time, then seems to reboot.

Wonder if something else is up with my device?

@PaulWieland?

JeffSteinbok, 2023-12-31:
Ok, so, quick update.  I flashed to 2.56 and now I can seem to control the door, but no status is reported back.  It always says it's closed, but the "open" button basically does the same thing as the remote button does.

JeffSteinbok, 2023-12-31:
Ok, confirmed now it's the same as the wall button issue.  CLosing

closed 2023-12-31 by JeffSteinbok