ratgdo/mqtt-ratgdo #30: Support for Chamberlain B4505T

falconindy, 2023-12-16:
Hi, recently received a version 2.52i board from you and installed firmware 2.54 in hopes of getting this working with my Chamberlain B4505T opener and Home Assistant. I'm quite sure I've connected everything properly, as MQTT correctly reports events from the opener as they occur but publishing commands to the opener via HA does nothing (both the door and the light). I've attached
[logs](https://github.com/ratgdo/mqtt-ratgdo/files/13693292/esp-web-tools-logs.txt) for you. Any thoughts on why this doesn't work? The B4505T has a yellow learn button and should be a 2.0 opener.

I will say that I don't fully understand how the opener is wired (done about 6 months ago by a local company) -- I've attached the harness for the red/white/black on the back of the opener to the ratgdo but have not wired anything else to the board. My current solution using OpenGarage just uses a hijacked security 2.0+ switch to actuate the door using the same red/white inputs on the opener.

FrankJaco, 2023-12-18:
I hope you don't mind answering a quick Noob question.  Where can I find the logs like the ones that you attached?

Thx,

fj

dbrewer-wdc, 2023-12-18:
@FrankJaco, if you go to the main ratgdo page, there is a link https://paulwieland.github.io/ratgdo/flash.html . 
Go through the motions like you are going to install FW on your ratgdo, but instead, after selecting your USB attached ratgdo, select the "logs and console".  After you capture what you want to see, you can save the log and share it.

atwoodjw, 2023-12-18:
Looks like you have client id = 0. See: https://github.com/ratgdo/mqtt-ratgdo/issues/23#issuecomment-1857304473

falconindy, 2023-12-20:
> Looks like you have client id = 0. See: [#23 (comment)](https://github.com/ratgdo/mqtt-ratgdo/issues/23#issuecomment-1857304473)

Thanks for triaging -- you're spot on. Works perfectly with firmware 2.55.

closed 2023-12-20 by falconindy