ratgdo/mqtt-ratgdo #31: Reed switches

gshepperd, 2023-12-17:
I need to use a Reed switch for my security 1 gdo. Is it possible to just have one Reed switch? (Closed is closed and not closed is open) it would be really difficult to install a second and I am having a hard time finding any kind of answer.

PaulWieland, 2023-12-17:
Two switches are required in order to determine the four door states of open, closing, closed and opening. Install the magnet at the top of the door. Put one reed switch near it when the door is shut, and the other back on the track so it is tripped when the door is fully open.

closed 2023-12-17 by PaulWieland

dbrewer-wdc, 2023-12-18:
@PaulWieland, for the serial display, can change the display to show the key and val when the key is x40? For my GDO that doesn't give door status, I so see x40 80 when the door is moving and x40 00 when it is stopped. With my one reed switch for when the door is closed, I determine all 4 states. 
if door reed switch closed, door is closed
if reed switch just opened and x40 status is x80, door is opening.
if door status was opening and now x40 status is x00, door is open.
if door status was open and now x40 status is x80, door is closing.
if the door status was closing and x40 status is x00, door is stopped.

The only status I don't have is if the door is stopped when opening. I'll just report it as open.
But at the same time, I don't think it matters to many people if the door reports door opening or closing. It is a temporary state anyway.