ratgdo/mqtt-ratgdo #35: Another Sec+ 1.0 Wall panel (041A6318) apparently not compatible

FrankJaco, 2023-12-20:
I have a Chamberlain HD900D opener with battery back-up and a purple learn button, that is connected to a 041A6318 smart wall panel. I am currently running Ratgdo 2.54 MQTT software. When the Panel is connected to the circuit all kinds of squirrely things happen. When it is disconnected Ratgdo seems to work as expected (assumingly in emulation mode); I can raise, lower and reverse direction of the door. turn the light on/off and read the state of the obstruction sensors, but that is the extent of what I can see and do.

But my wall panel is quite smart...  built in motion detector, temperature sensor, clock with ability to program lockout times etc. etc. And importantly it would warn me if my battery backup was getting low or has failed along with many other diagnostic codes. With the exception of the battery state and the diag, I likely could emulate many of the features of the panel with other devices/sensors and Home Assistant. But I much rather have a working panel. 

Will there be any updates to the firmware MQQT or ESPHome that may allow me to reconnect my panel? Or is it a lost cause?  I have an extra 041A6318 panel and would be willing to send it to you if it helps.

It is my understanding from a previous comment, that most of the intelligence is in the opener not the panel.  If my panel is a lost cause, will we eventually see firmware that will provide access to the backup battery health and the other diagnostic codes?

Thx for all, and do keep up the great work. With the MyQ situation going on, no doubt your plate is as full as it gets.

fj

PaulWieland, 2024-01-02:
Update to the latest firmware and then pull the serial log from ratgdo to confirm what's going on with your wall panel.

Sudo-Rob, 2024-01-03:
Same issue here.  I'll try to grab a serial log for you.

reslip1, 2024-01-05:
I am seeing a similar issue with Security+ 1.0 and Chamberlain model 459950

mwilgar, 2024-01-09:
To add to the list, my 398LM Liftmaster Smart Control Panel is also exhibiting the same problems that have been reported. Purple Button (Security + 1.0), MQTT, v2.57. Door acts as though the RATGDO doesn't know where it is - up/down, in some other house ;) HomeAssistant is confused and sometimes randomly opens the door in the middle of the night. Homekit (from HA is even worse)

I'd be happy to provide logs, but I'd need some instruction on how to do that (without Windows)

dstathos, 2024-01-09:
> To add to the list, my 398LM Liftmaster Smart Control Panel is also exhibiting the same problems that have been reported. Purple Button (Security + 1.0), MQTT, v2.57. Door acts as though the RATGDO [...]

+1 on 398LM wall panel.  Seeing same activity and willing to provide logs once I know how to.

dstathos, 2024-01-09:
I think this is the procedure to collect logs (found from another post here):

1.  Connect the usb port to a computer
2.  Open esp web tools and go to console and logs
3.  Click reset device to get a fresh log
4.  Try and command the door through mqtt
5.  Command the door from the wall panel
6.  Download and share the log

I will try to do this myself in the next few days

mwilgar, 2024-01-09:
> 2\. Open esp web tools

From my Googling, "ESP Web Tools" is part of ESPHome. Since this is a Security + 1.0 situation I'm on MQTT, so I don't have ESPHome in my Home Assistant. 

I feel like I'm missing something.

rlowens, 2024-01-09:
> I feel like I'm missing something.

"ESP Web Tools" is the "[Click here to launch the ratgdo ESP Web Tools flasher to install the firmware](https://paulwieland.github.io/ratgdo/flash.html)" link on [the main ratgdo page](https://paulwieland.github.io/ratgdo/).

https://paulwieland.github.io/ratgdo/flash.html then on the "MQTT...Choose version:" box pick anything and click CONNECT, select the correct port and Connect, then LOGS & CONSOLE.

mwilgar, 2024-01-11:
> "ESP Web Tools" is the "[Click here to launch the ratgdo ESP Web Tools flasher to install the firmware](https://paulwieland.github.io/ratgdo/flash.html)" link on [the main ratgdo [...]

Thanks rlowens, that was quite clear and very helpful!

I've attached my log on my original post.

mwilgar, 2024-01-11:
> To add to the list, my 398LM Liftmaster Smart Control Panel is also exhibiting the same problems that have been reported. Purple Button (Security + 1.0), MQTT, v2.57. Door acts as though the RATGDO [...]

[esp-web-tools-logs (1).txt](https://github.com/ratgdo/mqtt-ratgdo/files/13908389/esp-web-tools-logs.1.txt)

PaulWieland, 2024-01-11:
> [esp-web-tools-logs (1).txt](https://github.com/ratgdo/mqtt-ratgdo/files/13908389/esp-web-tools-logs.1.txt)

From the log:
`3A5C 3900 3707 WARNING: Digital wall control panel detected, but it is not requesting door status. Door status will be unavailable.`

You have an incompatible wall panel. See the FAQ "[ratgdo can’t detect the door status of my Security + 1.0 opener](https://paulwieland.github.io/ratgdo/09_faq.html)".

mwilgar, 2024-01-11:
> You have an incompatible wall panel. See the FAQ "[ratgdo can’t detect the door status of my Security + 1.0 opener](https://paulwieland.github.io/ratgdo/09_faq.html)".

Yup, that's what I was afraid of. Thanks MyQ, there goes another $110 (I need 2)

It might be worth adding the (growing) list of incompatible wall panels to the Wiki. 

The FAQ mentions the 889LM as an alternative - Amazon lists the 889LMMC as an updated version of the 889LM - is the 889LMMC known to work?

PaulWieland, 2024-01-11:
You can use the dry contact integration if you want, which is much cheaper. Get a pair of doorbell /momentary buttons and two sets of reed switches. All it for ~$30?

Also, that amazon listing is bogus. [The 889LMCC is the recalled panel](https://www.liftmaster.com/myq-control-panel/p/889LMMC).

Sudo-Rob, 2024-01-14:
Apparently, the recall associated with the 889LMMC panel only apples to units manufactured between March 2022 and October 2022.  Here's a link:

[https://support.chamberlaingroup.com/s/889LMMC-041A7928-3MC-warranty-replacement](https://support.chamberlaingroup.com/s/889LMMC-041A7928-3MC-warranty-replacement)

FYI, I just connected up one manufactured in May 2023 to a Sears Craftsman version of the Chamberlain HD900D.  So far, it seems to be working with MQTT v2.57 and is reporting "supported_features: 11."  Once I figure out how to save the log, I'll post it.