ratgdo/mqtt-ratgdo #38: Can bins be posted to the releases page?

jerrm, 2023-12-21:
This really applies to all three builds.

Currently there is no easy way to download the bin files (unless I'm missing something).

It would be great to be able to setup github watch alerts to see when a new release is posted.

avocadosalsa, 2023-12-21:
They are posted over in this repo: https://github.com/PaulWieland/ratgdo/tree/main/docs/flasher

jerrm, 2023-12-22:
Thanks. 

Still, it's not the most intuitive location.  I"d never think of drilling down under "docs" in the closed repo.

rlowens, 2023-12-22:
Can the Releases page here have a link to https://github.com/PaulWieland/ratgdo/tree/main/docs/flasher ?

sejgit, 2023-12-22:
the link on the units webpage for GitHub takes you to the "latest release". it has been pointing to 2.1 for a while

joshjohanning, 2023-12-23:
@jerrm It looks like the 2.56 binaries are posted to the releases on this repo now 😄 

https://github.com/ratgdo/mqtt-ratgdo/releases

closed 2023-12-24 by jerrm