ratgdo/mqtt-ratgdo #41: motion is detected but never clear & messes up light

sejgit, 2023-12-22:
you likely know this one and are working on it but I just figured out (v2.56) that motion is detecting and updating mqtt it just never clears.
related, when you walk into garage and motion turns the light on, it ends up making mqtt the opposite of reality.
ie. off=on

joshjohanning, 2023-12-23:
cc #36 

I think that's reflected as a todo in the readme:
> * [ ] Flip light status on when motion detected packet is sent

sejgit, 2023-12-25:
got it. thanks for all you are doing!

closed 2023-12-25 by sejgit

Goose66, 2024-01-10:
> I think that's reflected as a todo in the readme:
> 
> > * [ ]  Flip light status on when motion detected packet is sent

That TODO appears to be for updating light status on motion detected. This issue is related to the fact that the motion status is never set to cleared. Going through the code, it is clear that, while the code clears the motion flag itself, it is only reported when it is set to 1 ("detected") but never reported when it is reset to 0. What's more, it is a retained state, so the status in the MQTT broker will always return "detected" if it is queried or on new subscriptions, whether there is motion or not. 

IMO, it should be either 1) managed like the other statuses with a previousMotionState and clear and detected messages posted each time it changes, or 2) should not be retained with "detected" messages published only when motion detected (and thus we would never expect a "clear" status).