ratgdo/mqtt-ratgdo #42: Suggestion: Door open and close could possibly be more accurate after door is stopped

p-jean, 2023-12-23:
The closeDoor() and openDoor() functions will toggle the door if it is in the "stopped" state.

On the 8500, for example, and I suspect all models, the door always reverses direction when toggled after a stop, so it could be useful to remember what the door was doing before it was stopped.

In some cases, maybe two toggle commands will be required to get the desired result.

MYeager1967, 2024-01-20:
Pressing the stop button in HA simply reverses the door for me. Not sure if it's supposed to stop it where it is or if that's even possible with my door. I think I've done it with the wall button, but I can't swear to it. My wall button is a dumb version so that may affect it as well...

Goose66, 2024-01-20:
It depends on the opener (at least in dry contact mode). For me sending stop (equivalent to button press) stops the door on open but reverses the door on close.

MYeager1967, 2024-01-23:
I'm using the MQTT side of it on a security+ opener. Not sure how it's supposed to react and it doesn't really matter that much. Everything else works perfectly.

aureateflux, 2024-02-13:
There are a lot of different behaviors among dry contact GDOs, so maybe the best thing is to have a method for the user to configure garage door behavior?

For example, my garage door (pds/linear 800) has the following behavior: first button press initiates travel in a direction opposite to the last direction travelled (if it had closed, it will now open) *unless* the motor is currently running, in which case it will stop. The next press reverses direction of travel.

If ratgdo were configured to expect this behavior, either only one button would be needed in HA or it could grey out all non-relevant options (as it already does when an open or closed state is reported).  

This might also help deal with variations in how GDOs signal commands over dry contact: my GDO clearly just sends the same command each time regardless of direction of travel. And unlike @MYeager1967 and @Goose66 , my stop button in HA is completely nonfunctional.

A drop-down list categorizing each behavior, an interview -style "what happens when ratgdo does this?" wizard, or direct button mapping option might all work.

BootsSiR, 2024-04-18:
I ran into this just now.  When I park my car in my garage after a spirited drive, I close the door then stop it about 3/4 closed.  This lets the car cool off without making my garage 40 C.  I wrote a small piece of code that monitors the ratgdo and when the door is detected in a stopped position, it will automatically close it after 1h.  Only problem is, when I issue the close mqtt command the door will actually open since closing the door actually a toggle so the door reverses direction and opens up.  I'm going to start tracking the door direction in my code but if there's an easier way to manage this, please let me know.