ratgdo/mqtt-ratgdo #43: Sec+ 1.0 door status not reported

sangdrax8, 2023-12-27:
I just got two of these devices wired up and installed, running v2.56 for hardware 2.50.  At first I had issues until I realized that I needed to switch to the 1.0 protocol at the very bottom of the page, and things started working mostly.  I am looking at the MQTT Explorer and I see a status for 

- Availability
- light
- lock
- obstruction

but I don't see any status for door for either one of them.  Luckily, I can activate the door from both my existing well switch (it is smart with light sensors, battery health, etc) and from inside HA.  So I am going to be using this to see how things go.  Is there something else I am missing here about the status?  I thought it would either have detected the signal, or emulated and faked one.  I just don't know how the rest of my MQTT status results are showing up but not the door.

PaulWieland, 2023-12-27:
Please read the FAQ.

sangdrax8, 2023-12-28:
Sorry, I thought when I read FAQ there was a section about mine having status "supported" but with an *.  It mentioned something about listening to the communication or emulating if, so I thought either way it would send something.

I now see there is one that directly calls out (for anyone else reading this later)

```
Non myQ digital wall panels use digitial communication to control the door opener, but they do not ask the door opener for it’s status and so ratgdo has no way to know what the door status is. If you have one of these wall panels then there are two options:

  - Use ratgdo’s dry contact control and reed switches to get the door status
  - Buy either an Analog wall panel (a simple door bell/momentary contact switch will work), or a digital one which is myQ compatible such as the 889LM, which queries the door for it’s state.
```

So it would seem I don't have status and that is working as expected.  Sorry about the ticket IDK where I read the last info I got from.

closed 2023-12-28 by sangdrax8