ratgdo/mqtt-ratgdo #44: No Lock Entities in Homeassisant with Board 2.5/V2.56

restauffer, 2023-12-29:
I am running a 2.5 board with V2.56 firmware on a sec +1 gdo.  I am not seeing entities for status or command of the lock function in HA. Do I need to take steps to make the appear?

bubtheengineer, 2023-12-30:
It's not supported in the HA MQTT autoconfig yet.  You can manually add a retained MQTT entry to create it.
Topic is specific to your HA setup, but something like this.: homeassistant/lock/ratgdo/config

Update this to match the existing entries in MQTT for ratgdo:
`{
  "~": "ratgdo",
  "name": "Lock",
  "unique_id": "ratgdo-lock",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/lock",
  "payload_lock": "lock",
  "payload_unlock": "unlock",
  "state_locked": "locked",
  "state_unlocked": "unlocked",
  "state_topic": "~/status/lock",
  "device": {
    "name": "ratgdo",
    "identifiers": "ratgdo",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.56",
    "configuration_url": "http://1.1.1.1/"
  }
}`

restauffer, 2023-12-31:
@buntheengineer  - thank you!! That worked just as I wanted. Happy New Year!!!!

closed 2023-12-31 by restauffer