ratgdo/mqtt-ratgdo #45: Variable might be too small for value in transmit()

p-jean, 2023-12-31:
In transmit(), if millis() is nearly the same as lastRX, then the value of txDelayLen will be smaller than expected due to wrap-around with the uint8_t type.

```
void transmit(byte* payload, unsigned int length){
                ...
		uint8_t txDelayLen;
		byte tempPayload[1];

		txDelayLen = (lastRX + 275) - millis();
		delay(txDelayLen);
                ...
}
```

PaulWieland, 2024-01-02:
This isn't an issue.

closed 2024-01-02 by PaulWieland

edrikk, 2024-01-05:
Hi @PaulWieland 
Although probably not hit, his (base-)point is still valid.
That is:  millis() returns an "unsigned long" which is 32 bits ([definitely is on Arduinos](https://www.arduino.cc/reference/en/language/variables/data-types/unsignedlong/)) while uint8_t is an unsigned 8-bit integer.

It's not a bad idea to define txDelayLen (and any other variable that is working with/against millis() as "unsigned long" similar to  this [recent commit in the homekit firmware](https://github.com/ratgdo/homekit-ratgdo/commit/d5f410a452e1c24246a8682a528a56821ca1c30b).