ratgdo/mqtt-ratgdo #48: Security 2.0+/2.56 Only obstruction status reports and can only toggle door/light

dbavatar, 2024-01-01:
Assuming I'm reading the FAQ correctly, this should work. I'm using 880LM control panels with a Chamberlain MyQ Wifi 001D8353

There is no status of light or door, subsequently door and light toggles work backwards (open closes, etc)

There is status of the motion detection from the panel, but the motion is reported as obstruction, as well as the beam.

What debug can I do?

Logs:

[esp-web-tools-logs.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13802502/esp-web-tools-logs.txt)

dbavatar, 2024-01-02:
top of the log:

```
WiFi connected
Local IP: 192.168.3.201
SoftAP IP: (IP unset)
Server started
Webserver ready. Open http://ratgdo-5826.local/ in your browser
Launching webserver for improv
Starting ArduinoOTA service
doorCommandTopic: ratgdo-5826/command/door
lightCommandTopic: ratgdo-5826/command/light
lockCommandTopic: ratgdo-5826/command/lock
Using security+ 2.0
Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.56
IMPROVhttp://192.168.3.201
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: ratgdo-5826/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED
```

PaulWieland, 2024-01-02:
Double check your connections / post a picture of your wiring. It looks like the red/ctrl wire isn't connected to the GDO.

dbavatar, 2024-01-02:
[image]

Red seems OK, goes direct to opener, and the pass through to the panels which work.

This opener has a lead acid battery backup, and the battery status does get reported and displayed on the panels, perhaps there are some protocol differences?

dbavatar, 2024-01-02:
Additional weirdness, I thought this was a fluke, but tinkering with it now seems definitive. I have 2x 880LM LCD control panels on the system (although one is Chamberlain branded). One of the panels gets kicked off the bus whenever the ratgdo reboots (asks to press bar to resync). So I tried disconnecting it, then finally moving to a single 882LM "dumb" panel. Seems to make no difference to the ratgdo behavior above regardless?

PaulWieland, 2024-01-03:
An 882LM is still a digital panel. It uses sec plus 2 rolling code protocol to communicate with the door opener.

PaulWieland, 2024-01-03:
Try doing a reinstall with erase. This will force ratgdo to generate a new client id.

dbavatar, 2024-01-03:
Erase generated a new ID it looks like, but had no effect on missing status reports or the conflict with the one 880LM. Tried it twice.

exisiting client ID: 464539
...
exisiting client ID: 139539
...
exisiting client ID: 168539

PaulWieland, 2024-01-03:
Why do you say there is a conflict? 
Anyway I think theres a bad component on your board. Please send an email to ratgdo@gmail so we can arrange an exchange.

closed 2024-01-03 by PaulWieland

dbavatar, 2024-01-03:
Email sent.

Just a wild guess, anyway this is what happens when I soft-reset the ratgdo:

https://github.com/ratgdo/mqtt-ratgdo/assets/14057304/5143a76f-98eb-4fdc-9974-9a11fb51072a

dbavatar, 2024-01-08:
Status problem was a hardware fault, new unit fixed it.

Weird conflict with panel still a problem, I'll open another ticket with logs.