ratgdo/mqtt-ratgdo #51: ratgdo disconnects from mqtt server every 1 minute exactly

timespaced, 2024-01-04:
my ratgdo disconnects and reconnects from the MQTT server (tested with ratgdo-homebridge plugin and with Home Assistant) every 1 minute exactly.

I have done the following, amounting to many hours, and the issue persists:
- Removed ratgdo-homebridge plugin, deleted all files (persists and accessories folders) re-added plugin. Issue persists.
- Installed a second docker installation with a fresh homebridge, installed homebridge ratgdo plugin. Issue persists.
- Re-flashed ratgdo, erasing data, using the default firmware, not ESPHome Home Assistant.
- Installed VM of Home Assistant (HA). Installed MQTT and MQTT explorer. Connected newly re-flashed ratgdo to HA. MQTT Explorer shows the same thing the logs for the homebridge-ratgdo plugin shows - the ratgdo disconnects and reconnects every minute.
- I pinged the ratgdo for several minutes - each time the ratgdo disconnected from the MQTT server there was a corresponding drop of the ping (just 1 return of "request timed out" each time the ratgdo disconnects from the MQTT server).

timespaced, 2024-01-21:
@PaulWieland I cannot reply to you at the Square email, it errors.

Here is the ESPHome "Logbook" after flashing to ESPHome - issue persists
`ratgdov2.5i e50c9e Lock remotes was unlocked
13:27:04 - 18 seconds ago
ratgdov2.5i e50c9e Client ID changed to 7,443,769.0
13:27:04 - 18 seconds ago
ratgdov2.5i e50c9e Learn turned off
13:27:04 - 18 seconds ago
ratgdov2.5i e50c9e Paired Devices became unknown
13:27:04 - 19 seconds ago
ratgdov2.5i e50c9e Light turned off
13:27:04 - 19 seconds ago
ratgdov2.5i e50c9e Door was opened
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Dry contact light turned off
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Dry contact close turned off
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Dry contact open turned off
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Motor became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Button turned off
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Obstruction cleared (no problem detected)
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Motion cleared (no motion detected)
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Toggle door became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Sync became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Query openings became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Query status became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Safe mode boot became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Restart became unknown
13:27:03 - 19 seconds ago
ratgdov2.5i e50c9e Light became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Learn became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Door became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Paired Devices became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Lock remotes became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Dry contact light became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Dry contact close became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Dry contact open became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Motor became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Button became unavailable
13:27:02 - 20 seconds ago
ratgdov2.5i e50c9e Obstruction became unavailable
13:27:02 - 20 seconds ago
`

timespaced, 2024-01-21:
I just disconnected all of my Eero wifi APs and connected my Unifi Pro AP

Problem persists:
```
ratgdov2.5i e50c9e Motion cleared (no motion detected)
13:41:03 - 2 minutes ago
ratgdov2.5i e50c9e Toggle door became unknown
13:41:03 - 2 minutes ago
ratgdov2.5i e50c9e Sync became unknown
13:41:03 - 2 minutes ago
ratgdov2.5i e50c9e Query openings became unknown
13:41:03 - 3 minutes ago
ratgdov2.5i e50c9e Query status became unknown
13:41:03 - 3 minutes ago
ratgdov2.5i e50c9e Safe mode boot became unknown
13:41:03 - 3 minutes ago
ratgdov2.5i e50c9e Restart became unknown
13:41:03 - 3 minutes ago
ratgdov2.5i e50c9e Light became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Learn became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Door became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Paired Devices became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Lock remotes became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Dry contact light became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Dry contact close became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Dry contact open became unavailable
13:41:02 - 3 minutes ago
ratgdov2.5i e50c9e Motor became unavailable
13:41:02 - 3 minutes ago
```

timespaced, 2024-01-21:
My wifi SSID has special characters, so I thought maybe that could be it - changed to a single word, no spaces or special characters.  Issue persists.

PaulWieland, 2024-01-22:
```
QUEUE MSG SENT [ratgdo-89969/status/availability] 
online
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: ratgdo-89969/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
0
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: ratgdo-89969/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
1
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: ratgdo-89969/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED
```

This indicates an issue with the broker or network. It's connecting and immediately being disconnected, then connecting again.

timespaced, 2024-01-22:
> ```
> QUEUE MSG SENT [ratgdo-89969/status/availability]
> online
> ### MQTT DISCONNECTED ###
> Connecting to
> MQTT Broker...
> MQTT Last Will Params:
> willTopic: ratgdo-89969/status/availability
> willPayload: [...]

Pretty sure it’s not my wifi network as I think I’ve ruled that out above. I don’t see why or how it could be my router (pfsense) as the connections should be going device <-> device and never hit the router. 

Broker I assume is the server? I’ve tried 3 different brokers; homebridge MQTT via docker and a dedicated IP, and 2 flavors of brokers within Home Assistant (in a VM with a different IP): mqtt and ESPHome. So how could it be a broker issue?

Are you certain it’s not a hardware issue with the ratgdo? And if so, why are you certain?

PaulWieland, 2024-01-22:
>Are you certain it’s not a hardware issue with the ratgdo? And if so, why are you certain?

If it were a hardware issue, it wouldn't connect at all.

PaulWieland, 2024-01-22:
Have you tried the ESP Home firmware?

timespaced, 2024-01-22:
> Have you tried the ESP Home firmware?

Yes. Results are above. Same issue.

timespaced, 2024-01-22:
And here are the ratgdo ESPHome logs:

```
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:36:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:37:14
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:38:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:39:14
```

timespaced, 2024-01-22:
> Have you tried the ESP Home firmware?

Yes. Logs from the ESPHome side of Home Assistant are [above](https://github.com/ratgdo/mqtt-ratgdo/issues/51#issuecomment-1902770070). Same issue. 

And here are the ratgdo ESPHome logs:

```
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:36:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:37:14
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:38:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:39:14
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[I][ota:117]: Boot seems successful, resetting boot loop counter.
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:40:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:41:14
```

The line `[I][ota:117]: Boot seems successful, resetting boot loop counter.` seems interesting - anything there?

timespaced, 2024-01-22:
> Have you tried the ESP Home firmware?

Here's the entire log file from the ratgdo in case this is helpful:
```
rl l  |  l | d c|    { # c  oo lgg   cp  l{lrlp g  l  cg |   c  oo  l  d` gg  lor   g{{ `x ' r      cn | c  oo   l` oo d o{   gs$` r  n{l`  pd`  o d[I][logger:351]: Log initialized
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[D][number:012]: 'Rolling code counter': Sending state 61.000000
[D][ratgdo:335]: Set opening duration: 0.0s
[D][ratgdo:341]: Set closing duration: 0.0s
[D][number:012]: 'Client ID': Sending state 218425.000000
[C][switch.gpio:011]: Setting up GPIO Switch 'Status door'...
[D][switch:016]: 'Status door' Turning OFF.
[D][switch:055]: 'Status door': Sending state OFF
[D][switch:016]: 'Status door' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Status obstruction'...
[D][switch:016]: 'Status obstruction' Turning OFF.
[D][switch:055]: 'Status obstruction': Sending state OFF
[D][switch:016]: 'Status obstruction' Turning OFF.
[C][light:035]: Setting up light 'Light'...
[D][light:036]: 'Light' Setting:
[D][light:041]:   Color mode: 
[D][binary_sensor:034]: 'Motion': Sending initial state OFF
[D][binary_sensor:034]: 'Obstruction': Sending initial state OFF
[D][binary_sensor:034]: 'Button': Sending initial state OFF
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: AC:0B:FB:E5:0C:9E
[D][wifi:061]: Loaded saved wifi settings: Got eem!
[D][wifi:455]: Starting scan...
[D][binary_sensor:034]: 'Dry contact close': Sending initial state OFF
[D][binary_sensor:034]: 'Dry contact open': Sending initial state OFF
[D][binary_sensor:034]: 'Dry contact light': Sending initial state OFF
[D][ratgdo_secplus2:463]: Collision detected, waiting to send packet
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_OPENINGS
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: ALL
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: REMOTE
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: KEYPAD
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: WALL_CONTROL
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: ACCESSORY
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[W][component:214]: Component ratgdo took a long time for an operation (0.12 s).
[W][component:215]: Components should block for at most 20-30ms.
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][number:012]: 'Rolling code counter': Sending state 62.000000
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[... 1006 lines cut ...]
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:40:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:41:14
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:42:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:43:14
[... cut ...]
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:44:13
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 16:45:16
```

PaulWieland, 2024-01-22:
Can you create a DHCP reservation on your network using a different IP address? It looks like something is kicking it off of your network.

timespaced, 2024-01-22:
> Can you create a DHCP reservation on your network using a different IP address? It looks like something is kicking it off of your network.

Added static mapping in pfSense.  While I was there I added a firewall rule to allow all traffic from the ratgdo new IP and from HomeAssistant, just in case.

Issue persists.

```
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:20:19
;l d  |  d < l #|    r c b  oo lgg   cp l{d;lp o  d  cg <   c  gg  d  d` 'n l gs   n{r `8 n r ܜ   cg | c  gg   d` gg  lo{   g{l` r  n{l`  xl`  o l[I][logger:351]: Log initialized
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[D][number:012]: 'Rolling code counter': Sending state 427.000000
[D][ratgdo:335]: Set opening duration: 0.0s
[D][ratgdo:341]: Set closing duration: 0.0s
[D][number:012]: 'Client ID': Sending state 218425.000000
[C][switch.gpio:011]: Setting up GPIO Switch 'Status door'...
[D][switch:016]: 'Status door' Turning OFF.
[D][switch:055]: 'Status door': Sending state OFF
[D][switch:016]: 'Status door' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Status obstruction'...
[D][switch:016]: 'Status obstruction' Turning OFF.
[D][switch:055]: 'Status obstruction': Sending state OFF
[D][switch:016]: 'Status obstruction' Turning OFF.
[C][light:035]: Setting up light 'Light'...
[D][light:036]: 'Light' Setting:
[D][light:041]:   Color mode: 
[D][binary_sensor:034]: 'Motion': Sending initial state OFF
[D][binary_sensor:034]: 'Obstruction': Sending initial state OFF
[D][binary_sensor:034]: 'Button': Sending initial state OFF
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: AC:0B:FB:E5:0C:9E
[D][wifi:061]: Loaded saved wifi settings: Got eem!
[D][wifi:455]: Starting scan...
[D][binary_sensor:034]: 'Dry contact close': Sending initial state OFF
[D][binary_sensor:034]: 'Dry contact open': Sending initial state OFF
[D][binary_sensor:034]: 'Dry contact light': Sending initial state OFF
[D][ratgdo_secplus2:463]: Collision detected, waiting to send packet
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_OPENINGS
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: ALL
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: REMOTE
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: KEYPAD
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: WALL_CONTROL
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:213]: Query paired devices of type: ACCESSORY
[W][ratgdo_secplus2:430]: Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[W][component:214]: Component ratgdo took a long time for an operation (0.12 s).
[W][component:215]: Components should block for at most 20-30ms.
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][number:012]: 'Rolling code counter': Sending state 428.000000
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[D][ratgdo_secplus2:466]: Collision detected, waiting to send packet
[... 998 lines cut ...]
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:21:29
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:22:29
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:23:29
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:24:29
[... cut ...]
[I][ota:117]: Boot seems successful, resetting boot loop counter.
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:25:29
[W][api.connection:102]: Home Assistant 2023.12.4 (10.30.18.244): Connection reset
[D][api:102]: Accepted 10.30.18.244
[D][api.connection:1121]: Home Assistant 2023.12.4 (10.30.18.244): Connected successfully
[D][time:044]: Synchronized time: 2024-01-22 18:26:29
```

PaulWieland, 2024-01-22:
Is this with it powered directly off of your computer's USB port?

timespaced, 2024-01-22:
> Is this with it powered directly off of your computer's USB port?

Currently plugged into a powered laptop dock.   But I've had the issue plugged directly into the front USB port of my PC as well (as recently as this morning).  Also the issue was occurring when it was hooked up to the GDO and plugged directly into the wall with the provided USB cable and brick.

I just switched out the provided kit USB cable with another one and same issue persists.

timespaced, 2024-01-28:
@PaulWieland Both the Homebridge docker and Home Assistant VM are hosted on my Unraid server.  I thought I would remove the Unraid server from the equation to see if the problem persits - I installed Homebridge OS on a raspberry pi 3B and the problem persists - continuously disconnects every 1 minute.  Moreover, pinging the device results in a dropped packet at the exact 1 minute mark when the ratgdo disconnects from the mqtt server.

At this point I have removed the possibility of it being 1) wifi iusse, as I have changed out my wifi device and isolated the wifi network so only the ratgdo was connected, 2) it is not a homebridge/mqtt issue, as the issue persists across homebridge mqtt, homeassisstant mqtt, and homeassistant esphome, 3) removed the possibility of the issue being related to the USB cable or port as I have tried several cables, ports, the included kit charging brick, and multiple computers, 4) related to the broker host as I have placed the broker server on multiple different physical computers with different IP addresses as well.

I'm not sure what else there is to check on my side, maybe its time for an exchange of the ratgdo?

timespaced, 2024-02-03:
Issue was DHCP lease time. It was set to 61 seconds for some reason. Changed it to default 86400 and the issue is resolved. Thanks for your help!

closed 2024-02-03 by timespaced