ratgdo/mqtt-ratgdo #52: doesn't work with Genie 7055 opener, causes lockout of panel as well

changes13, 2024-01-05:
Tried with v1 and v2 security.  In both cases, the motor locks out any operation, even from the panel.  ratgdo operations dont work.

Panel: [image]
motor: https://www.geniecompany.com/garage-door-opener-/model-7055
log with v1 sec : [7055-v1.txt](https://github.com/ratgdo/esphome-ratgdo/files/13836491/7055-v1.txt)

PaulWieland, 2024-01-05:
Genie openers are not chamberlain security plus. You have to use the dry contact wiring and control protocol.

closed 2024-01-05 by PaulWieland

changes13, 2024-01-05:
I changed the config to dry contact.  Same behavior.  Is there something else I should try?