ratgdo/mqtt-ratgdo #53: Only publish HA discovery messages for supported functions

Goose66, 2024-01-05:
I am using the ratgdo devices to control Genie door openers through Dry Contact mode. Even though my Genie Control Pads have separate light and lock switches, these are not (yet) supported. Accordingly, my ratgdo can only control the door. Accordingly, it would be helpful if the ratgdo didn't publish HA discovery message for the light, so that my integration wouldn't create a light node. Similarly, if there was some way for the ratgdo to know if the obstruction sensors are not connected (mine are too high voltage), then it would also be great if the ratgdo would not publish an HA discovery message for the obstruction sensors.

PaulWieland, 2024-01-06:
I'll have it skip the auto discovery messages for light / lock when using the dry contact control protocol

Goose66, 2024-01-06:
That would be great!

Is there any (straightforward) way for it to detect whether the obstruction sensors are connected in dry contact control mode and skip that discovery message as well? In other words, if no obstruction sensor connected, don't publish this one:
```
{
    "~": "ratgdons/ratgdo-26944",
    "name": "Obstruction",
    "unique_id": "ratgdo-26944_08:3A:8D:FA:6B:FB_obs",
    "availability_topic": "~/status/availability",
    "device_class": "motion",
    "state_topic": "~/status/obstruction",
    "payload_on": "obstructed",
    "payload_off": "clear",
    "device": {
        "name": "ratgdo-26944",
        "identifiers": "ratgdo-26944_08:3A:8D:FA:6B:FB",
        "manufacturer": "Paul Wieland",
        "model": "ratgdo",
        "sw_version": "2.56",
        "configuration_url": "http://192.168.1.101/"
    }
}
```
I know this would probably take more work than just skipping the "Light" discovery message in dry contact control protocol, but would be helpful in my  integration with Universal Devices controllers, in which control nodes are created based on the presence of these discovery messages (and I would assume Home Assistant as well).