ratgdo/mqtt-ratgdo #54: Any chance of support for non MyQ openers?

ATaylor5596, 2024-01-06:
Not a bug but more of a request. I foolishly bought a Ryobi opener many years ago since the app is now dead and never truly worked in the first place, I figured I could use ratgdo since it's a security+ 2.0 opener (I think). Well of course I was half wrong, even though it's rolling code/security+ 2.0 it apparently uses different commands as pressing the toggles in Home Assistant will cycle through the add-on accessories, the light and eventually open and close the door. Is there any hope of possibly decoding the commands so it could work? If it's explained how I could send whatever logs or anything that might be needed to do that, if not I also understand I was mainly curious.

Goose66, 2024-01-06:
My Genie CM7600IC works well in dry contact control mode for door control and status (the obstruction sensors are too high voltage and aren't connected). However, the Genie wall control keypads have light and lock buttons that work (even with the ratgdo attached) using some sort of signaling that I feel like could be implemented in the ratgdo firmware. I imagine the signaling in this case is probably far more primitive than MyQ security protocol, but I think it could be replicated if I sort of knew what to look for and/or knew what kind of measurements that were needed. I just am not sure how to contribute here.

Also, if this needs to be a new/different issue, than feel free to move or let me know.

PaulWieland, 2024-01-06:
Can you share the serial log from ratgdo?

1. connect the usb port to a computer
2. Open esp web tools and go to console and logs
3. Click reset device to get a fresh log
4. Try and command the door through mqtt
5. Command the door from the wall panel
6. download and share the log

ATaylor5596, 2024-01-07:
So, I hope I did this correctly, this is the log from trying to control the garage with mqtt. An interesting note, call number 86 and 119 closed the door successfully, call 112 opened it, call 89 turned on the light and call 92 changed the fan speed. I'm sure a few others turned on the parking lasers and locked the keypad in vacation mode but I didn't catch when those happened. Hopefully I did the logs correct for at least the open and close to work. As an aside the obstruction sensors work fine without issue.
[mqtt control.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13852304/mqtt.control.txt)

PaulWieland, 2024-01-07:
Yeah that’s kind of wild that it responds to anything. 

Next step would be to let ratgdo log each of the commands sent by your wall panel and gdo. The serial log isn’t ideal for this because it’s hard to keep track of which physical event belongs with which codes, but if you can manage I’ll take a look and see if there’s a discernible pattern in the data. 

Basic things that need to be figured out:

* door control command
* Light control command
* door status report [open, closing, closed, opening]
* Light status report [on, off]

the other commands and status if there are any can be sorted out later.

ATaylor5596, 2024-01-07:
So, strange thing, I must be doing something wrong because neither the wall keypad, the garage remote or the car remote report any codes being sent in the logs. I'm not sure how that's possible though. Granted the indoor keypad is wireless like everything else, but it still should be throwing out codes, but the logs aren't picking up anything. Seems like I'll need a flipper zero unless it's actually user error getting the logs.

rlowens, 2024-01-07:
> Granted the indoor keypad is wireless like everything else

Why would the wire see any codes if all the remotes you are using are wireless?

Also, what model of GDO is this exactly? Got a link to the manual? What are the terminals you connected to the ratgdo meant for if the wall control is wireless?

ATaylor5596, 2024-01-07:
So, the indoor wall keypad is wireless and sends commands to a receiver unit that is connected to the terminals on the garage door unit. I was under the assumption that codes would be sent from the keypad to the receiver then passed to the unit via the terminal connection, or regular commands would be sent to the receiver, encoded and then sent to the unit. I thought that was the case because dry contact mode and other dry contact units didn't work except for once but I chocked that up to a fluke since I couldn't get it to happen repeatedly. But if the logs aren't showing anything coming from the car/spare remotes or keypad/receiver then I'm clearly mistaken about how this works in some form or another. As for the other questions, the model of the unit is the Ryobi GD201, the keypad is GDA401. As for manuals, I included the .pdf for the keypad and the opener itself. The terminals on the ratgdo itself are connected to the receiver and passed through to the door unit which works fine.
[keypad.pdf](https://github.com/ratgdo/mqtt-ratgdo/files/13855125/keypad.pdf)
[door unit.pdf](https://github.com/ratgdo/mqtt-ratgdo/files/13855142/door.unit.pdf)

Like I said, I understand this was created specifically to work with MyQ units and not others, but I figured if I want something to work with my unit, I will have to figure it out and build it myself and/or ask for assistance from those who know more than me.

PaulWieland, 2024-01-07:
> So, strange thing, I must be doing something wrong because neither the wall keypad, the garage remote or the car remote report any codes being sent in the logs. I'm not sure how that's possible [...]

Not weird at all, it’s probably a totally different control protocol.

PaulWieland, 2024-01-07:
> Why would the wire see any codes if all the remotes you are using are wireless?

Because sec+ gdos report their status on the serial line.

ATaylor5596, 2024-01-08:
> Not weird at all, it’s probably a totally different control protocol.

Yes you're correct, it turns out rolling code and security+, which is apparently also rolling code are different things unbeknownst to me, I though all rolling code was the same. So it's using a completely different protocol for the codes and frequency for the remotes I presume. Do you have any idea how I could did that out of see what it's reporting on it's wires? I assume since it has/had a smart phone app and it has a program button, it has to be reporting something back in some way shape or form. If MyQ can be reverse engineered so should be able to of I can figure out how.