ratgdo/mqtt-ratgdo #56: Add genie light & lock controls

PaulWieland, 2024-01-06:
My Genie CM7600IC works well in dry contact control mode for door control and status (the obstruction sensors are too high voltage and aren't connected). However, the Genie wall control keypads have light and lock buttons that work (even with the ratgdo attached) using some sort of signaling that I feel like could be implemented in the ratgdo firmware. I imagine the signaling in this case is probably far more primitive than MyQ security protocol, but I think it could be replicated if I sort of knew what to look for and/or knew what kind of measurements that were needed. I just am not sure how to contribute here.

Also, if this needs to be a new/different issue, than feel free to move or let me know.

_Originally posted by @Goose66 in https://github.com/ratgdo/mqtt-ratgdo/issues/54#issuecomment-1879745164_

Goose66, 2024-01-09:
The good news is the control signaling is not hard at all to figure out and no digital oscilloscope is required. The bad news is, it's a simple resistor ladder that is (AFAIK) going to be hard to interface with for both sensing and controlling. Below is a (bad) circuit diagram. If any EEs have any thoughts, let's hear it.
[image]

VladimirKuletski, 2024-06-06:
Up.
Would love to have this feature onboarded!