ratgdo/mqtt-ratgdo #58: LiftMaster 8500 Jackshaft Door will not open or close

dwdonnelly, 2024-01-07:
Rec’d my ratdgo 2.5i kit – many thanks.  
Home Assistant (HA) – ratgdo will not open or close or stop garage door.
Flashed ratgdo using web interface for my mqtt broker.  All went well.  Connected ratgdo to my LiftMaster JackShaft Model 8500 garage door opener. Verified all wiring, and verified solid wire connections in terminals. 
Original garage door opener wiring:
[image]

Mqtt broker has added ratgdo.  It shows as a device in mqtt. 
HA auto discovered ratgdo.  Verified all LiftMaster 8500 garage door opener remotes work, external keypad and indoor push button all function normally, as does the obstruction sensor.
Verified on HA, when obstruction sensor is blocked, HA displays not clear.  When obstruction removed, HA shows cleared.
 
[image]

[image]

 
But – select up or down arrow or stop square on HA….nothing happens.
 
[image]

(Note: I have a LiftMaster 8500 jackshaft model and there is no light on this garage door.)

Since all remotes and wall switch work, this – to me – verify the wiring is correct.
Took logs of attempts to open and close garage door via HA/ratdgo. 

```
QUEUE MSG ARRIVED [garage_doorratgdo-garage/command/door] 
open
MQTT: open the door
rolling code for 88|door1 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 88|door2 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
Write successful
QUEUE MSG ARRIVED [garage_doorratgdo-garage/command/door] 
close
MQTT: close the door
rolling code for 89|door1 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 89|door2 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
Write successful
QUEUE MSG ARRIVED [garage_doorratgdo-garage/command/door] 
stop
MQTT: stop the door
The door is not moving.QUEUE MSG ARRIVED [garage_doorratgdo-garage/command/door] 
close
MQTT: close the door
rolling code for 90|door1 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 90|door2 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
Write successful
{l l��| �l�|�l�c|����s�#�c��og�$g'���bx��${$sd8�o��l��co�<���#��gn� l��l`�'o�ln{���o{;�`x�'�;������c'�|�#��gn� �l`�no�dorǛ�gr$`{��o{d`���lsl ��'�d
Mounting LittleFS...
[setup.json]
JSON parsed
Storage OK, restoring WiFi and MQTT config.
DPsoftware domotics
Connecting to: 
"myWifi"...
Using static IP address
Set WiFi output power to: 20.50
.
WIFI CONNECTED
IP Address: xxx.xxx.x.xx
nb of attempts: 1
IMPROVhttp://xxx.xxx.x.xx�

WiFi connected
Local IP: xxx.xxx.x.xx
SoftAP IP: (IP unset)
Server started
Launching webserver for improv
Starting ArduinoOTA service
doorCommandTopic: garage_doorratgdo-garage/command/door
lightCommandTopic: garage_doorratgdo-garage/command/light
lockCommandTopic: garage_doorratgdo-garage/command/lock
Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.1
Syncing rolling code counter after reboot...
rolling code for 91|reboot1 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 92|reboot2 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 93|reboot3 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 94|reboot4 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 95|reboot5 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
rolling code for 96|reboot6 : 5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxXXXXXX
Write successful
IMPROVhttp://xxx.xxx.x.xx�
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: garage_doorratgdo-garage/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [garage_doorratgdo-garage/command/#] 
TOPIC SUBSCRIBED [garage_doorratgdo-garage/set_code_counter] 
QUEUE MSG SENT [homeassistant/cover/ratgdo-garage/config] 
{
  "~": "garage_doorratgdo-garage",
  "name": "ratgdo-garage",
  "unique_id": "ratgdo-garage_xx:xx:xx:xx:xx:xx",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "ratgdo-garage",
    "identifiers": "ratgdo-garage_xx:xx:xx:xx:xx:xx",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1",
    "configuration_url": "http://xxx.xxx.x.xx/"
  }
}
QUEUE MSG SENT [homeassistant/light/ratgdo-garage/config] 
{
  "~": "garage_doorratgdo-garage",
  "name": "ratgdo-garage Light",
  "unique_id": "ratgdo-garage_xx:xx:xx:xx:xx:xx_light",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/light",
  "payload_on": "on",
  "payload_off": "off",
  "state_topic": "~/status/light",
  "device": {
    "name": "ratgdo-garage",
    "identifiers": "ratgdo-garage_xx:xx:xx:xx:xx:xx",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1",
    "configuration_url": "http://xxx.xxx.x.xx/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/ratgdo-garage/config] 
{
  "~": "garage_doorratgdo-garage",
  "name": "ratgdo-garage Obstruction",
  "unique_id": "ratgdo-garage_xx:xx:xx:xx:xx:xx_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "ratgdo-garage",
    "identifiers": "ratgdo-garage_xx:xx:xx:xx:xx:xx",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1",
    "configuration_url": "http://xxx.xxx.x.xx/"
  }
}
QUEUE MSG SENT [garage_doorratgdo-garage/status/availability] 
online
Obstruction status clear
QUEUE MSG SENT [garage_doorratgdo-garage/status/obstruction] 
clear

```

Your help in this matter to resolve this issue and get the garage door to function is greatly appreciated.

PaulWieland, 2024-01-07:
[8500 is sec plus 1.0. ](https://github.com/ratgdo/mqtt-ratgdo/wiki)

closed 2024-01-07 by PaulWieland

dwdonnelly, 2024-01-08:
Paul, am a bit confused by your reference.: [8500 is sec plus 1.0.](https://github.com/ratgdo/mqtt-ratgdo/wiki).  In reading this https://paulwieland.github.io/ratgdo/, I am using version 2.5 and have the 889LM wall control panel.  Need some clarity.  What should I do or not do or ?  What am I missing here?  Thanks in advance for your understanding and help.

rlowens, 2024-01-08:
Did you set the ratgdo as Sec+ 1.0 mode?

PaulWieland, 2024-01-08:
Sorry I linked to the wrong page. I fixed the link.

dwdonnelly, 2024-01-08:
Corrected link - thanks.  However, same problem.  I reflashed ratgdo, went back through the setup.......but in the configuration process - there is nowhere to set the security protocol.  I do not get this ability:
Control Protocol - Required.
Chamberlain / LiftMaster with YELLOW learn button (Except some wall mounted jackshaft openers): Choose Security + 2.0
Chamberlain / LiftMaster with PURPLE or RED learn button (And some wall mounted jackshaft openers): Choose Security + 1.0
Other openers - choose Dry Contact

this requirement does not show up.

dwdonnelly, 2024-01-10:
Here is a screen capture of my configuration.  Nowhere to select security.
[image]

How should I proceed?  Is this a bug in the system?

PaulWieland, 2024-01-10:
I dont know what firmware you flashed, but that's not the right one. There's no OTA upload in the header and there's no protocol selector at the bottom. You need v2.57 and you need to choose security + 1.0 control protocol in the configuration page.

[image]

dwdonnelly, 2024-01-10:
Here is what I chose - seems to be the same as what you present....yes?  
[image]

Am I in the correct url?

I can reflash and screen shoot at each step - would that help?

dwdonnelly, 2024-01-11:
Long story – very short.  Reflashed ratgdo.
  
All went very well.  On the ratgdo web page, YES! There is a Control Protocol selection.  I MISSED THIS (MY BAD – BIG TIME).  I saw and focused on firmware update. (tunnel vision)

However, I selected Security +1, and then updated the firmware to V2.1.  System updated and rebooted.  However, the garage door would not respond.
 
So – reflashed, install complete, selected security +1, DID NOT update firmware to V2.1 and ratgdo works.

Firmware Update:
I selected ratgdo_esp8266_v2.1.bin
 
The update description would seem to indicate I need to update as I am using Home Assistant.
Or is “esp” for ESPHome vice mqtt?

My humble apologies - your guidance is appreciated.

wesmoc, 2024-01-28:
@dwdonnelly when you say "ratgdo works", are you saying you were able to control the garage door (open/close), get door status, and get obstruction status?

I have the same opener and have only been able to get obstruction status, and I am eager to get this functioning.

dwdonnelly, 2024-01-28:
@wesmoc  thanks for the feedback and excellent that you got it to work!