ratgdo/mqtt-ratgdo #59: web config fails to load if mqtt server is down.

ipha, 2024-01-07:
After moving my mqtt server to a new machine, I tried to login to the web config to update the mqtt settings, but couldn't. Web config never loaded, but it still responded to pings.

Tried a hard reboot -- no change.

Turned my old mqtt server back on, gave it another reboot, and I was able to get back to the settings.

Running 2.57 on v2.5i hardware.

edrikk, 2024-01-08:
Yah the MQTT setup / behaviour on boot is a super weak point of the current implementation.

An MQTT server might literally catch fire and not be restartable.  

Regardless, if the user doesn’t know about this issue, they would never think to give the new MQTT server the same IP as the old.  They would just try and fail to fix the IP on the ratgdo.

IDEALLY FQDN would also be allowed, but at a minimum the ratgdo side should absolutely not depend on the MQTT server being up to allow someone to at least access the UI.

PaulWieland, 2024-01-08:
Will be addressed in v2.58