ratgdo/mqtt-ratgdo #60: Strange conflict with 880LM panel

dbavatar, 2024-01-09:
This panel keeps getting knocked off when ratgdo is attached, previously discussed in #48. Seems to work fine without it.

I don't know if the log is capturing anything relevant, but here it is.

[esp-web-tools-logs (4).txt](https://github.com/ratgdo/mqtt-ratgdo/files/13867296/esp-web-tools-logs.4.txt)

[image]

PaulWieland, 2024-01-09:
https://github.com/ratgdo/esphome-ratgdo/issues/140

closed 2024-01-13 by dbavatar

martinecker, 2024-04-02:
First off let me say how awesome of a project ratgdo is. Thanks for doing this!

I'm getting the same thing as the original poster with my Liftmaster 880LM wall digital panel, and others are reporting the same issue in this thread: https://community.home-assistant.io/t/installed-a-ratgdo-now-getting-error-on-my-wall-control/

I've verified my wiring is good with a multimeter and checking continuity. The panel has worked without getting disconnected for years before. Also, just removing power from the ratgdo but leaving it wired in, the panel always works. As soon as the ratgdo is powered and sits between the opener and the wall panel this happens within an hour or two. I can then push and hold the wall button for a few seconds and the panel starts working again for a few hours.

Seeing https://github.com/ratgdo/esphome-ratgdo/issues/140 I'm willing to buy a new 880LM panel for $50, but it does seem like there's some other issue going on here.

dbavatar, 2024-04-05:
I have to say buying a new panel did get rid of the message - but I don't think it fixed the problem. Often, the panel doesn't respond on the first or even second button press, I believe it's just hiding it's status.

martinecker, 2024-04-30:
Interesting. Thanks for reporting back.

I haven't purchased a new wall panel yet, but have replaced my 20 year old panel with another 10 year old one that I was able to get my hands on for free.
It did not have the issue, but I noticed that it had the panel motion sensor that turns on the opener light disabled. Once I enabled it the problem came back. Disabling the motion sensor the wall panel has been stable for several days now.
I haven't put back my older 20 year old panel yet to see if it would work similarly with the motion sensor off (it's been on this whole time).

General consensus in other issues/threads appears to be that the capacitors on these old wall panels need to be replaced, which I may do. Alternatively, getting a new wall panel fixes the problem as well (probably because of brand new capacitors).