ratgdo/mqtt-ratgdo #62: Can't configure a static IP address on ratgdo

pfurrie, 2024-01-09:
Version: 2.56 and 2.57

When configuring ratgdo the first time, it picks up a DHCP address of 10.15.15.235.  Then I enter a new address in the IP Address field.  Click the "Save Config & Reboot" button and...

When I'm prompted to log back on to the ratgdo device, it is still using the original DHCP-acquired address.  The specific address I'd given was ignored.

Am I missing something?  Is that IP address field not meant for setting an address, just showing whatever address is currently in-use?

NukeThemTillTheyGlow, 2024-01-09:
I had a similar problem with static IP addresses.  Seemed like it worked the first time and then every other time I made a chance (reflashing, updating config settings) no matter what I did in the unifi network app it stayed with the old IP.  The two ESPHome ratgdo's it was a one and done setting and they are fine.  Something tells me its a DHCP release thats needed, but not sure how to do that from the ratgdo, guessing you have to get to the command promp somewhere.

xAcrosonicx, 2024-01-12:
Personally, I prefer setting DHCP addresses within my router - I have found it to be the most reliable (as well as shows all static IPs in one place). 

Are you able to configure yours from your router?

pfurrie, 2024-01-12:
Yes, I can, but that's beside the point.  The docs for ratgdo say the IP
address can be set (non DHCP) but it can't do what it claims.

xAcrosonicx, 2024-01-12:
> Yes, I can, but that's beside the point. The docs for ratgdo say the IP address can be set (non DHCP) but it can't do what it claims.
> […](#)

I was curious back when I commented on whether or not it would work for me, so I unset my DHCP and used the web interface for the ratgdo to set a different IP and rebooted - worked like a charm for me. For reference, I am running v2.57 on a v2.5 board , so not sure why it wouldn't be working for you. Hopefully Paul has some insight for you

PaulWieland, 2024-01-12:
Where does it say that?

> 
> On Jan 11, 2024, at 9:32 PM, Pat Furrie ***@***.***> wrote:
> 
> Yes, I can, but that's beside the point. The docs for ratgdo say the IP
> address can be set (non DHCP) but it can't do what it claims.

pfurrie, 2024-01-12:
On the GitHub IO website, the page concerning configuration::

https://paulwieland.github.io/ratgdo/02_configuration.html

It says:

   - IP Address - Will be filled in automatically with the DHCP address,
   but can also be set manually.

PaulWieland, 2024-01-12:
Thanks. I will correct it. 

> On the GitHub IO website, the page concerning configuration::
> 
> https://paulwieland.github.io/ratgdo/02_configuration.html
> 
> It says:
> 
> - IP Address - Will be filled in automatically with the DHCP address,
> but can also be set manually..

closed 2024-01-23 by PaulWieland

pfurrie, 2024-01-24:
Bummer.  Was this not possible to do?

mattlqx, 2024-07-30:
Docs still say this is possible. I'd really like to not use DHCP as I keep my IoT devices on a different subnet.