ratgdo/mqtt-ratgdo #66: Lock / Unlock not showing in HA for MQTT on Sec 1.0

beit-no1, 2024-01-14:
Hi there,
I just received my Ratgdo ( 2 pcs) this past week and have configured them as MQTT with Home Assistant.  I tried at first with Esphome, but I have the purple learn button and SEC 1.0 so I had to go back to MQTT for now.  I flashed it and it is v2.57 firmware on the 2.5i hardware ([ratgdo v2.57 for hardware v2.50])

I have the open / close as well as control over the light and it shows if the sensor is clear... Everything appears to be working, but there is no lock / unlock the remotes.  Is there some way to add this within Home Assistant that maybe I'm unaware. It does not show as an entity under the Device.

[image]

rlowens, 2024-01-14:
Duplicate of https://github.com/ratgdo/mqtt-ratgdo/issues/20 and https://github.com/ratgdo/mqtt-ratgdo/issues/44

Will be added in a new release. For now, you can use instructions at https://github.com/ratgdo/mqtt-ratgdo/issues/44#issuecomment-1872602768

Goose66, 2024-01-28:
Look at the "homeassistant" topic in your MQTT Explorer. Are the "lock" device autodiscovery messages there? Looks like these were added in the 2.57 firmware release.

closed 2024-02-07 by PaulWieland