ratgdo/mqtt-ratgdo #68: 889LM flashing red and yellow on ratgdo command

FeatherKing, 2024-01-16:
Hi, I have 4 ratgdo installed, 1 is security 2.0 and 3 are security+. Everything is working on emulation mode, but I just moved one of the security+ wall panels from a doorbell to an 889LM to test it out (emulation mode worked for all doors)

When i hooked up the 889LM it flashes quickly red+yellow (manual indicates this is chargin it battery). then the light goes solid yellow. During this time the ratgdo correctly shows all status updates in home assistant/mqtt. Wall panel works, wireless remote works. When I issue commands over mqtt, the wall panel flashes slowly red+yellow (manual indicates this wall panel isnt compatible with my opener). After a few reboots of the wall panel, it eventually goes solid yellow and everything works again.

I can go back to emulation mode, but I expected this to work? Logs are attached.
[garage-south-889LM-log.txt](https://github.com/ratgdo/mqtt-ratgdo/files/13944932/garage-south-889LM-log.txt)

FeatherKing, 2024-01-16:
i wasnt sure how to add a good timestamp in the logs, so near the end where there is a lot of looped output, i waited for a full cycle:
```
1    3A4C 3900 3A4C 3855  | door state: 55 1010101
2    3A4C 3900 3A4C 3855  | door state: 55 1010101
3    3A4C 3900 3A4C 3855  | door state: 55 1010101
4    QUEUE MSG ARRIVED [ratgdo-garage-south/command/light] 
5    off
6    MQTT: turn the light off
7    3A4C 3233 33FE 3131 Door state syncing
8    QUEUE MSG SENT [ratgdo-garage-south/status/door] 
9    syncing
10    3131 313F 3131 3131 3535 3333 3855  | door state: 55 1010101
11    Door state closed
```

the time between 6 and 7 was at least 5 minutes. the wall panel reset 2 or 3 times before it finally went solid yellow. I can get another log tonight. let me know if you want me to try any other steps or a different sequence. In my testing yesterday this was always constant:
```
the wall panel would be solid yellow
send a command through HA/MQTT
the log output would print the command
the opener would perform the command
but HA/MQTT topic would never get the result of the command
wall panel would flash red+yellow
```

FeatherKing, 2024-02-03:
Just moved to the esphome secplus 1 to see if that would help and i still have the same behavior here. The logging through the esphome build doesnt seem to be quite as detailed as what i was pulling of the serial console. Status is all reported fine, obstruction, open/close, light, but if i issue a command the wall panel reboots a few times.

FeatherKing, 2024-02-03:
log output from esphome
```
18:18:58	[D]	[ratgdo:095]	
Door state=CLOSED
18:18:58	[D]	[ratgdo:204]	
Light state=OFF
18:18:58	[D]	[ratgdo:210]	
Lock state=UNLOCKED
18:18:59	[D]	[light:036]	
'Light' Setting:
18:18:59	[D]	[light:047]	
  State: ON
18:18:59	[D]	[ratgdo:210]	
Lock state=UNLOCKED
18:21:29	[D]	[ratgdo:095]	
Door state=CLOSED
18:21:30	[D]	[ratgdo:204]	
Light state=ON
18:21:30	[D]	[ratgdo:210]	
Lock state=UNLOCKED
18:21:30	[D]	[ratgdo:204]	
Light state=ON
18:21:30	[D]	[ratgdo:210]	
Lock state=UNLOCKED
18:21:30	[D]	[ratgdo:204]	
Light state=ON
18:21:30	[D]	[ratgdo:210]	
Lock state=UNLOCKED
18:21:31	[D]	[ratgdo:095]	
Door state=CLOSED
18:21:31	[D]	[ratgdo:204]	
Light state=ON
18:21:31	[D]	[ratgdo:210]	
Lock state=UNLOCKED
18:21:32	[D]	[ratgdo:204]	
Light state=ON
18:21:32	[D]	[ratgdo:210]	
```

wall panel was rebooting for about two minutes before the log output started again

Snuffy2, 2024-02-11:
I spent some time trying to grab some logs to help troubleshoot this as I believe I'm having the same issue.
Opener: Craftsman 139.53975SRT1
Panel: 889LMMC
ratgdo: 2.5i (MQTT 2.57)

## Sequence of events:
The door takes approx 13 seconds to open and also to close

### Power cycled ratgdo (changed from wall power to laptop)....

### MQTT Open 1: [esp-web-tools-logs - MQTT Open 1.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233361/esp-web-tools-logs.-.MQTT.Open.1.txt)
* Panel beeped 3x
* ~1 min until syncing
* ~1:30 sec to open (triggered by light off?)

### MQTT Close 1: [esp-web-tools-logs - MQTT Close 1.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233363/esp-web-tools-logs.-.MQTT.Close.1.txt)
* Panel beeped 2x
* 1:40 from close to light off

### MQTT Open 2: [esp-web-tools-logs - MQTT Open 2.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233366/esp-web-tools-logs.-.MQTT.Open.2.txt)
* Panel beeped 1x
* 37 sec to syncing
* 42 sec to open

### MQTT Close 2: [esp-web-tools-logs - MQTT Close 2.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233367/esp-web-tools-logs.-.MQTT.Close.2.txt)
* Panel beeped 2x
* 37 sec to syncing
* 1:20 to close

### Power Cycle Garage Door (ratgdo remained on): [esp-web-tools-logs - GDO Power Cycle.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233368/esp-web-tools-logs.-.GDO.Power.Cycle.txt)
* Panel beeped 1x
* 25 sec to syncing

### MQTT Open 3: [esp-web-tools-logs - MQTT Open 3.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233369/esp-web-tools-logs.-.MQTT.Open.3.txt)
* Panel beeped ~7x
* 1:10 to syncing
* (1:43 light off but not in logs)
* 3:45 sec to open

### MQTT Close 3: [esp-web-tools-logs - MQTT Close 3.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233371/esp-web-tools-logs.-.MQTT.Close.3.txt)
* 13 seconds to closed

### Power Cycle ratgdo: [esp-web-tools-logs - ratgdo Power Cycle.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233372/esp-web-tools-logs.-.ratgdo.Power.Cycle.txt)

### MQTT Open 4: [esp-web-tools-logs - MQTT Open 4.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233373/esp-web-tools-logs.-.MQTT.Open.4.txt)
* Panel beeped 2x
* 40 sec to syncing
* 1:20 to open

### MQTT Close 4: [esp-web-tools-logs - MQTT Close 4.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233376/esp-web-tools-logs.-.MQTT.Close.4.txt)
* (I missed the initial MQTT close command in the logs)
* Panel beeped >10x
* 1:10 to syncing
* (1:40 light off but not in logs)
* 5:35 to closed

### Panel Open: [esp-web-tools-logs - Panel Open.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233378/esp-web-tools-logs.-.Panel.Open.txt)
* 13 seconds to open

### Panel Close: [esp-web-tools-logs - Panel Close.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233379/esp-web-tools-logs.-.Panel.Close.txt)
* 13 seconds to close

### MQTT Open 5: [esp-web-tools-logs - MQTT Open 5.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233380/esp-web-tools-logs.-.MQTT.Open.5.txt)
* 13 seconds to open

### MQTT Closed 5: [esp-web-tools-logs - MQTT Close 5.txt](https://github.com/ratgdo/mqtt-ratgdo/files/14233381/esp-web-tools-logs.-.MQTT.Close.5.txt)
* Panel beeped 2x
* 38 seconds to syncing
* 1:20 to closed

**Note:** It would be **very** helpful if a timestamp or some other counter could be added to the logs

geophilips, 2024-02-20:
I can second this. A couple of days ago, I switched out my old wall panel for an 889LM. After charging, it worked for a bit, then started randomly opening and closing, with the incompatible lights flashing. Finally, it stopped midway, at which point I disconnected the power and reconnected my old panel.

Unfortunately, since I only had a little bit of time, I couldnt get any logs (also my ratgdo wont reconnect after a reboot without me toggling between channels 6 and 11 on my 2.4ghz network.)

Following this issue therefore. thanks!

FeatherKing, 2024-07-19:
still having this issue on 2024.7.0 esphome and a build date of today. Im hoping maybe this PR will help https://github.com/ratgdo/esphome-ratgdo/pull/304

```
16:27:49 | [D] | [ratgdo:203] | Lock state=UNLOCKED
16:27:49 | [D] | [ratgdo:197] | Light state=OFF
16:27:49 | [D] | [ratgdo:203] | Lock state=UNLOCKED
16:27:50 | [D] | [ratgdo:088] | Door state=CLOSED
16:27:50 | [D] | [ratgdo:197] | Light state=OFF
16:27:50 | [D] | [ratgdo:203] | Lock state=UNLOCKED
16:27:50 | [D] | [light:036] | 'Light' Setting:
16:27:50 | [D] | [light:047] | State: ON
16:27:50 | [W] | [ratgdo_secplus1:264] | [41136] Discard incomplete packet: [39 ...]
16:27:50 | [D] | [ratgdo:203] | Lock state=UNLOCKED
16:27:51 | [D] | [ratgdo:197] | Light state=ON
16:27:51 | [D] | [ratgdo:203] | Lock state=UNLOCKED
16:28:19 | [I] | [safe_mode:041] | Boot seems successful; resetting boot loop counter
16:28:26 | [D] | [ratgdo_secplus1:299] | wall panel is starting
16:28:26 | [D] | [ratgdo_secplus1:299] | wall panel is starting
16:28:27 | [D] | [ratgdo_secplus1:299] | wall panel is starting
```