ratgdo/mqtt-ratgdo #71: Door Open/Close Status Shows UNKNOWN in Home Assistant

joltman80, 2024-01-22:
SEC+1 Chamberlain HD900D (Purple Learn Button)
Wall Panel: Chamberlain 041A6318
RATGDO Board rev 2.5i
FW: 2.57

I've just installed the RATGDO on my GDO.  I've wired the obstruction sensors, wall panel and the ground directly to the RATGDO.  I've used the included small header with 3 wires to run to RED/WHITE/BLACK on the GDO.  I've got MQTT (Mosquitto Broker) installed in Home Assistant Core 2024.1.5.  All software packages are up to date.

The RATGDO is not reporting door status (OPEN or CLOSED).  In HA, the device shows a status of UNKNOWN for the door.  If the GD is closed, and I click the DOWN button in HA, the door opens.  If the door is opening or closing and I click the STOP button in HA, the door does not stop.  MQTT Explorer on my laptop shows the RATGDO as

* availability = online
* light = off
* lock = unlocked
* obstruction = clear

I can provide a log but I'm not sure how to get it from the MQTT firmware on the device.  Thanks for the help!

joltman80, 2024-01-22:
Looks like this is a duplicate of [Issue 35](https://github.com/ratgdo/mqtt-ratgdo/issues/35).  I've just made an online purchase of an 889LM as a replacement.  I'll run some new doorbell wire while I'm at it.  Feel free to close this as a duplicate.

PaulWieland, 2024-01-23:
Likely issue is your wall panel being digital and not querying for status. You can try emulation mode by unhooking your wall panel. Also in a future release I’m going to inject the query command for wall panels which don’t send the right query. The ESP Home firmware does this and it works in some situations.