ratgdo/mqtt-ratgdo #72: Lack of support

grouts112, 2024-01-23:
I received my ratgdo 2.5i on Dec 20th,
After the christmas chaos, I tried to flash the unit, however it fails.
I can successfully flash D1mini's with esphome and make bluetooth proxies, so it is not my cable or driver issue.
The esp12f is getting very hot, too hot to touch. I believe the unit to be faulty.
I have sent reply emails to the reciept for my order multiple times, and not received one word of assistance.

I have finally surcummed to my impatience and ordered a replacement ratgdo
As I have had zero response to my numerous emails about the first one I received which appears to be faulty!
I would very much appreciate any input from your side on the matter and if you can help with the replacement at all,
Extremely disappointed, and now I have to buy the “kit” as the board itself as I originally purchased is no longer listed to order. I really did not need the additional expense nor do I need a US power pack, pretty useless in AUSTRALIA!
I CERTAINLY HOPE THIS ONE WORKS and I have not just wasted more money on what appears to be a completely unsupported product.
I do understand and appreciate the effort of providing and shipping pre-made hardware product to help some of us implement your hard work developing a myQ solution, it was for that reason I wanted to purchase from you in the hope that you might profit from your work, however the lack of any response to my emails is incredibly poor and disappointing.

PaulWieland, 2024-01-23:
What email address did you order from?

grouts112, 2024-01-23:
Hi Paul,
 Email used is 'glen@routley.homeip.net'