ratgdo/mqtt-ratgdo #74: Motion state MQTT message is retained message

Goose66, 2024-01-28:
Since motion state is intended to be an "event," i.e. only "detected" will be sent on motion and never "clear," can the retained message on the motionStatus message sent in  sendMotionStatus() be sent with the retainedmessage flag set to "false" instead of "true?" That way, when a integration is restarted or forced to reconnect/resubscribe to the MQTT topics, it won't pickup the last motion "event" that probably occurred some time ago.

This was discussed as an alternative in another issue, but the issue got closed without having a chance to discuss or consider.

sejgit, 2024-01-29:
upvote for this one please.

haffi78, 2024-03-03:
Vote here