ratgdo/mqtt-ratgdo #76: unable to re-flash RATDGO

makebede77, 2024-02-10:
I've had mine (ratdgo v2.5i) installed and connected to HomeAssistant for few weeks. its been working perfectly. Suddenly it stopped showing me status and was unable to control my garage door. I took it off the garage attempting to reflash it, however its not even showing up as a connected device. I'm using the same laptop i had when i originally flashed it and drivers are installed. When I connect any other ESP32 device to the laptop i can see it. I've tried different cables (rated for data transfer). 

What do i need to do?

Thank you in advance.

rlowens, 2024-02-10:
Sounds like it died. Is there any visual signs of damage on the board (blown chips, etc)? With USB connected, do you get 3.3V between ground (the shield on the ESP8266 module) and the VCC pin on the bottom-left of the module?
[image]

makebede77, 2024-02-10:
I dont see any visual damage, at least for my untrained eyes. Testing between ground and VCC i'm getting .74.Am I sol?

Thank you!!

rlowens, 2024-02-11:
I'm not sure what kind of warranty period @PaulWieland offers, but it sounds like something died on the board. Just to verify, you were testing the voltage with nothing else connected to the board, right?

If you have a 3.3v power supply around, you could try powering it from that and see if it comes up then.

makebede77, 2024-02-11:
Correct, nothing else was connected to the board when i tested.
Unfortunately I dont have a 3.3v power supply.

@PaulWieland I am curious if any warranty is offered on the board.

rlowens, 2024-02-11:
I believe the method he has directed others on purchase issues is to reply to the order confirmation email, if you want to try that.

makebede77, 2024-02-13:
I did, I'm awaiting on a response.

Thank you!!