ratgdo/mqtt-ratgdo #78: Add option to firmware to force 802.11g (2.4ghz band)

aureateflux, 2024-03-01:
TLDR: If possible, add a compatibility option to force the ratgdo to use 2.4ghz band only (or force 802.11g) to resolve connection issues on some network configurations.

I was having intermittent connection issues with my new Ratgdo after setting it up with my network information. It took a long time to connect, but randomly would connect fairly quickly after a power cycle, only to randomly disconnect later.

After some digging, I found [this thread](https://www.reddit.com/r/homeassistant/comments/18i24is/ratgdo_wont_connect_to_wifi/) that matched my experience.

It appears this is caused by the esp8266 having difficulty negotiating the proper band with the router/access point in cases where the router supports having the same SSID for both 2.4ghz and 5ghz networks. It keeps getting handed the 5ghz band instead of the 2.4ghz band, which the esp8266 doesn't support, leading to failure after failure to connect.

Someone offered a solution for the ESPhome firmware that allows you to edit a config file to force it to use 802.11g, which forces it to use the 2.4ghz band.

Since my ratgdo is attached using dry contact, I am using the MQTT firmware, but I couldn't find a way to make a similar edit. I hit on the idea of using a wireless mac filter rule on the router to force the router to reject the ratgdo on the 5ghz bands. This works, and I've had solid up-time with it since.

Is it possible to add an option in the firmware configuration to force 802.11g on the Ratgdo or otherwise force it to use 2.4ghz in this network configuration? An on-device solution would be better than a cumbersome router/firewall rule, generally speaking, and would knock out a lot of bad user experience scenarios.

haffi78, 2024-03-03:
I think this is not a problem, since this ESP8266 does not support 5Ghz and there for does not get any frames from 5ghz and cannot and will not try to connect to that band.
Its the client desision to connect to which ever wifi and roam and such. So even if the SSID has both enabled, Devices that do not support 5GHz will not see the 5ghz. If your having troubles, check signal strength, check other 2.4 Ghz that might cause interferance, check which speeds(Rates)  your are supporting, Lower rates slow down your wifi if you have devices on them however they go further distance. I have mine on a Dual band , with out problems, I have a Cisco Wifi and the Access point is like 2 meters aways from the RatGDO.

aureateflux, 2024-03-03:
Applying a Mac filter to blacklist the ratgdo from joining a 5ghz network
resolves this connectivity issue. Forcing the esp8266 to use legacy mode
(forcing 802.11g) resolves this issue. If you were correct, these
workarounds would have no effect.

In the case of a network that has 5ghz and 2.4ghz on the same SSID, it can
see the SSID because it sees the SSID broadcast on the 2.4ghz band. But
it's not requesting the band, it's requesting the SSID.

haffi78, 2024-03-03:
Sorry that just not how beacons works, they are only send per Radio for its band, i.e. 2.4 send just becond for 2.4ghz and not 5 ghz, a Simple read on RFC and 802.11 will show you that.
This one for example.
What ever your problems and others are its not related to 5GHZ band, 
Forcing to use 802.11g is complete different then blocking 802.11n( N does not give you 5ghz unless its on that band which in this case its not ).
My RatGDO connects fine using 802.11n like I stated before.

The ESP8266 has a built in WiFi interface that supports 802.11 b/g/n 2.4 GHz networking. 5GHz frequencies are not supported.

https://en.m.wikipedia.org/wiki/IEEE_802.11#802.11n

https://www.7signal.com/news/blog/controlling-beacons-boosts-wi-fi-performance

aureateflux, 2024-03-03:
I'm sure you're correct, and I'm glad your RatGDO connects fine. I'll definitely look at my router's band steering settings.

But it doesn't change the fact that people have run into this issue with multiple RatGDOs and that the two workarounds I mentioned have worked with devices that are affected.

Instead of telling me how I'm wrong, why not focus on using your expertise to figure out what's actually happening? Maybe it would lead to a better solution than forcing the chip to use an older protocol. If there's nothing to be done, maybe something could be added to the readme to assist people in navigating the problem.

What's important is that it's addressed somehow, because as-is there are a lot of people running into it and just deciding RatGDO doesn't work.

haffi78, 2024-03-04:
Sorry if I am coming on to strong, just wanted to point this in the right directions, I would need more info to be able to help with debuging, but BandSteering could be and issue, thats when the APS send more beacons faster for the 5GHZ band, and sometime leave other stuff out of the beacon that causes trouble. Most of the time its something on the Access Points. But then sometime Access points dont have alot of settings depending on vendor.