ratgdo/mqtt-ratgdo #82: Home assistant is confused about the state of the door

brandoncc, 2024-03-24:
Hello, thank you for all of the hard work that goes into the ratgdo hardware and firmwares!

Home assistant is confused about the state of my door, even though I have installed it the way I understand the wiring diagram to be prescribing. My installation is for a Genie 7055, so I'm using dry contact and MQTT.

When I say it is confused, I mean that I have to say "open" to _close_ the door and vice versa. Also, it just says "opening" or "closing", it doesn't ever get to the "open" or "closed" state in home assistant. The first time I opened it, I hit the stop button in home assistant and then closed it from halfway. Is it possible that doing this has confused it? It seems like even if that happened, it would have worked itself out pretty quickly with the reed switches. I've tried opening/closing ~10 times and also restarted the ratgdo more than once since but it hasn't helped.

My board is a 2.53i, and the pins referenced below are the ones in this screenshot:

[image]

Here are the voltages at the pins during the different states:

| State | "close" pin | "open" pin |
|-|-|-|
| Door is open | 0v | 3.3v |
| Door is closed | 3.3v | 0v |
| Door is opening | 0v | 0v |
| Door is closing | 0v | 0v |

PaulWieland, 2024-03-25:
Those voltages are all backwards. 

When the door is closed, the closed limit switch should also be closed which pulls it to ground bringing the voltage to 0.

brandoncc, 2024-03-25:
Okay, I wondered about that. So when the door is opening or closing both "open" and "close" should have 3.3v?

The way I wired these is with NO/NC reed switches that have NO connected. That way they would close when the magnet is in proximity. Then the GND pin goes to the common screw on both reed switches.

brandoncc, 2024-03-25:
I switched my "NO" connections to "NC" and things seem to be working as I would expect. I guess I misunderstood "normally", which doesn't surprise me because I haven't installed an NO/NC contact in about 10 years and even then I have only installed a couple of them in my lifetime.

Please let me know if it sounds like I still have anything incorrectly wired. Thanks for your help!

closed 2024-03-25 by brandoncc

JoBrad, 2024-03-31:
@PaulWieland Any chance you can update the wiring diagram to note that reed switches should be wired in the NC position? I'm sure I'm not the only person who thought it indicated the NO position should be used. I switched to the NC position today, and this seems to have cleared up my issues with door status.