ratgdo/mqtt-ratgdo #83: Request option to disable emulation

abuker, 2024-03-25:
After power outages for my jackshaft 1.0+ door openers, my doors enter an uncontrollable state where they open randomly until the ratgdo is restarted.  I have tested this with MQTT and ESPHOME firmware, currently on ESPHOME.  I believe the ratgdo is booting faster than the 889LM panel, causing it to fall into emulation mode.  The panel then comes online, and falls into a loop of flashing yellow and red lights, and randomly opening the doors.  Is there an option to disable emulation mode, or set a longer timeout before it is enabled?

FireTime, 2024-06-23:
+1 on this. Had a power outage this morning and discovered my garage door opening and closing by itself. For those of us with 889LM wall panels we never need the emulation feature.

xxcite81, 2024-07-10:
+1 this as well. There could be a programmed timeout to wait so that after a power outage the door just doesnt randomly open. Was at work and came home to an open door. After opening it, the door began to open and close constantly without stopping almost like it wanted to self destruct. i flipped the breaker to get it to stop. i had to unplug the ratgdo then power the door, wait then plug the gdo in. The door resumed normal behaviour.