ratgdo/mqtt-ratgdo #85: Wireless keypad causes Ratgdo/Home Assistant to go out of sync (Sec 1.0+ w/ purple learn button)

FrankJaco, 2024-04-14:
I have a Chamberlain HD900D opener and a purple learn button, a 041A6318 smart wall panel. I am currently running Ratgdo 2.57 MQTT software.  My control panel is disconnected as it causes the many issues as reported elsewhere. I am using Smart Buttons and HA to open/close my door from within my garage, a typical chamberlain wireless remote from my car and have a typical chamberlain wireless keypad outside of my garage for those times I don't have my cellphone in hand.

During the several months that I have had the Ratgdo, I have had maybe 3 or 4 times it went out of sync and the status of the door is not updated and I cannot operated it via Home Assistant.  I previously attributed this to power blinks etc.   But I think I have identified an issue when using the chamberlain wireless **keypad**.  On those very rare occasions I use the **keypad** to open or close the door, the status is not picked up by Ratgdo and shown properly in HA.   Even a restart of HA doesn't resolve this issue. However if I use a wireless remote or the keypad again to close or open the door again so that its real status matches what is reflected in Home Assistant, I can once again operate the door via Ratgdo/HA without rebooting HA or the Ratgdo.

Are others experiencing this?  It seems very easy to test.   Is this just me or is there a bug in the MQTT firmware where if the door is toggled by the keypad the status is not picked up?

steinauf85, 2024-06-07:
Not sure if my issue is the same, but I was having trouble getting the door to close today and after trying with the wireless keypad it definitely seemed to get out of sync. I've got the same HD900D as you, with MQTT. wall panel is 041A7928-3MC which is a little different. its so out of sync that when it's all the way up it still think its closed and triggering it tries to get it to open even further, which strains the motor for a few seconds before it gives up. have had a few power blips due to recent storms this week, so also was wondering if that affected it. 

i'll take your theory into account when im troubleshooting and respond with any findings

FrankJaco, 2024-06-08:
@steinauf85

Thanks for the effort, but I have since switched to the ESPHome firmware and so far I haven't had an issue even when using the Chamberlain keypad multiple times. You may want to consider taking the ESPHome plunge as well.  (Note that your results could vary as you have a wall panel in the mix, where I have none installed.)

As mentioned in above post,  I do own a  041A6318 smart wall panel but I cannot have connected with either ESPHome or MQTT.  No stability whatsoever when it is wired in. Right now I am emulating all of the functions of the smart panel with Home Assistant smart buttons and sensors (except the status of my GDO battery health).

good luck!

fj