ratgdo/mqtt-ratgdo #87: intermittent door responses

scotmerrick, 2024-05-06:
Have 3 ratgdos and they work well for the most part.  More frequently, the open or close commands do not cause motion of the doors.  Seems like after power failure or upgrades to my Universal Devices Polisy plugin controller (Goose66 author), I lose door status  ("unknown") and the commands fail to open or close door.

Goose66, 2024-05-06:
What "mode" (dry contact, security+ 2.0, security+ 1)?

scotmerrick, 2024-05-07:
The Ratgdo boards use v2.57 security +1.0, 2.0 & dry contactSHM

Goose66, 2024-05-07:
No, you said you had 3 ratgdos controlling three garage door openers that are having the problem. What mode(s) are you using with your three garage door openers.

scotmerrick, 2024-05-07:
Sorry. What are you asking for?SHM

All are the sameSHM

Goose66, 2024-05-07:
Let's see if I can make this clearer: when you setup your ratgdo for your garage door openers, what operation mode did you select ("sec+ 2.0", "sec+ 1", or "dry contact")?

scotmerrick, 2024-05-07:
The firmware is 2.57 and the card is setup as “dry contact”

 

SHM