ratgdo/mqtt-ratgdo #90: Lost control on 2 of the 4 RatGDO units

skykep, 2024-07-07:
2 of my 4 units are on Craftsman belt-driven overhead units.  Flashed with 2.57 MQTT firmware and set for security + 2.0 (yellow buttons).  Worked for a bit, but today I discovered that these two Craftsman units are no longer responding to commands.  They still report status of door and light (although very delayed), and the wall-mounted door openers still work fine.

PaulWieland, 2024-07-30:
Try and reflash with full erase to see if a new randomly generated client ID fixes the problem.

skykep, 2024-07-30:
> Try and reflash with full erase to see if a new randomly generated client ID fixes the problem.

Yep, did that and it works for a few days, but ultimately ends up losing control again.

skykep, 2024-08-01:
Any way I can pull logs off it to help?  Just seems like it's losing the rolling code for control because the status hasn't stopped working at all.

PaulWieland, 2024-08-01:
I dont think its losing the rolling code, its just using a bad seed. There is some difference in the way the random rolling code is initialized between the esphome version and this version that I have to look into.

Until I figure it out, the only way to get a new random code with the mqtt firmware is to do a full erase and reflash.

Alternatively you can enable MQTT support in the esphome version and use that instead.

skykep, 2024-08-01:
Oh, that's some good news.  I'll give that a shot for now and report back.

closed 2024-10-09 by PaulWieland