ratgdo/mqtt-ratgdo #91: ratgdo keeps going offline

angrycatmeowmeow, 2024-07-22:
ratgdo 2.53i on mqtt firmware 2.57, dry contact.

The unit will not stay online. It shows unavailable in HA and is unreachable via its IP. There is nothing in my router logs about it disconnecting, and it still shows connected in the router interface. The only thing that will bring it back online is rebooting my entire mesh. It will eventually come back online but sometimes it's in seconds and sometimes 10+ hours. Other ESP devices stay connected.

What I've tried:

- Binding it to the closest AP
- MAC blocking to keep it from trying to join the 5ghz band
- Changed my wifi channel from auto to static
- Changed wifi channel width to 20/40
- Setting a static IP
- Bypassing Adguard Home and sending DNS to a regular DNS provider
- Moved the unit about 10ft away from anything else

angrycatmeowmeow, 2024-07-28:
Actually, it seems like restarting my mosquito broker will get it back online, too.

PaulWieland, 2024-07-28:
Do you have more than one device using the same mqtt client id? What does your broker log say?

angrycatmeowmeow, 2024-07-28:
If you the device name is the same as the client ID then it's the only one using the name I've given my ratgdo. I discovered restarting Mosquitto fixes it accidentally so I didn't get to check logs. I'll have to wait for it to break again.

PaulWieland, 2024-07-30:
I just re-read your original question and see you are using HA.... in which case I would strongly suggest you switch to [esphome-ratgdo](https://ratgdo.github.io/esphome-ratgdo/).

angrycatmeowmeow, 2024-07-30:
I thought esphome was not compatible with dry contact openers? I have a craftsman that's like 25 years old.

PaulWieland, 2024-07-30:
> I thought esphome was not compatible with dry contact openers? I have a craftsman that's like 25 years old.

[image]

angrycatmeowmeow, 2024-07-30:
Outstanding. I'll definitely load up the esphome firmware at the first opportunity.

angrycatmeowmeow, 2024-07-30:
Is it OK to load firmware while it's connected to the GDO? I was just gonna put my laptop on a ladder and flash it.

angrycatmeowmeow, 2024-07-30:
FYI this is all I have in my logs, it just went offline.

```
2024-07-30 12:28:17: Saving in-memory database to /data//mosquitto.db.
2024-07-30 12:58:18: Saving in-memory database to /data//mosquitto.db.
2024-07-30 13:28:19: Saving in-memory database to /data//mosquitto.db.
2024-07-30 13:58:20: Saving in-memory database to /data//mosquitto.db.
2024-07-30 14:15:53: Client garagedoor closed its connection.
```

angrycatmeowmeow, 2024-07-30:
So I installed ESPhome and adopted it but I'm getting this in the ESP logs and ratgdo logs attached.

```
======================== [SUCCESS] Took 146.15 seconds ========================
INFO Successfully compiled program.
INFO Connecting to 192.168.50.83
INFO Uploading /data/build/ratgdov25i-fada07/.pioenvs/ratgdov25i-fada07/firmware.bin (565616 bytes)
INFO Compressed to 386958 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 12.79 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.50.83 using esphome API
WARNING Can't connect to ESPHome API for ratgdov25i-fada07 @ 192.168.50.83: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.50.83', port=6053))]: [Errno 113] Connect call failed ('192.168.50.83', 6053) (SocketAPIError)
INFO Trying to connect to ratgdov25i-fada07 @ 192.168.50.83 in the background
```

[esp-web-tools-logs (2).txt](https://github.com/user-attachments/files/16432480/esp-web-tools-logs.2.txt)

angrycatmeowmeow, 2024-08-17:
Switched to esphome firmware and opened issue there.

closed 2024-08-17 by angrycatmeowmeow