ratgdo/mqtt-ratgdo #93: Unresponsive every 3-4 months

neldonado, 2024-07-23:
I am using the ratgdo 2.53 with security 1 garage door opener. I am using the mqtt with mosty good results, however I am running into the same issue every 3-4 months. I can't seem to figure out what's causing it, maybe others are facing similar issues and can chime in. But This is the third time in about a year where my garage door will start opening and closing every minute. The garage door opening becomes unresponsive and I have to unplug the controller and reflash the firmware. After this I am good for another 3-4 months.

skykep, 2024-07-29:
Not the same issue, but somewhat similar in where mine will stop controlling.  Still reads the status fine but unable to control 2 Craftsman belt-driven openers.  They work for a couple days and then quit.  The two wall-mounted Liftmaster units on my overhead doors still control fine.

Paul has not responded to any of my emails over the last month since I installed these.  Very frustrating to give someone money and they ghost their own product.

neldonado, 2024-07-29:
> Not the same issue, but somewhat similar in where mine will stop controlling.  Still reads the status fine but unable to control 2 Craftsman belt-driven openers.  They work for a couple days and [...]

Just out of curiosity, what's your Wi-Fi set up like? part of me is wondering if the device is getting bounced around to my different access points and just dropping out somehow but I haven't been able to figure out or reproduce it. It just will randomly happen. the first time, it opened up my garage door at 1 o'clock in the morning and set off my whole Home alarm .

PaulWieland, 2024-07-29:
@neldonado what control protocol are you using?

skykep, 2024-07-29:
> Just out of curiosity, what's your Wi-Fi set up like? part of me is wondering if the device is getting bounced around to my different access points and just dropping out somehow but I haven't been [...]

I have an access point within 10 feet of these units, so it's not changing access points.  I haven't had it open or close my doors randomly, fortunately.  If that were the case, I would rip these out immediately and demand a fix or a refund.  I'm trying to be patient for a fix because at least the status reporting is working fine.

skykep, 2024-07-29:
> @neldonado @skykep what control protocol are you using?

MQTT.  Supposedly this isn't so much of an issue for those using ESPHome.

PaulWieland, 2024-07-29:
> > @neldonado @skykep what control protocol are you using?
> 
> MQTT. Supposedly this isn't so much of an issue for those using ESPHome.

Threats not a control protocol. It’s either sec +1, 2 or dry contact.

neldonado, 2024-07-29:
> @neldonado @skykep what control protocol are you using?

@PaulWieland security +1

skykep, 2024-07-29:
Security + 2.0 for all four of mine.  Two of them work fine.

PaulWieland, 2024-07-29:
> security +1

What firmware version?

neldonado, 2024-07-29:
> > security +1
> 
> 
> 
> What firmware version?
> 
> 
> 
> 
> 
> 
> 
> 

V2.57 for hardware v2.50

PaulWieland, 2024-07-30:
skykeps issue is completely unrelated to the original reported issue. Deleting that part of the thread to clean this up, as there is already a separate open issue.

PaulWieland, 2024-07-30:
It only happening after months is really odd. I wonder if it's a capacitor failure in your wall control panel which is causing your wall panel to brown out & reboot. What model is your wall control panel?

neldonado, 2024-07-30:
> It only happening after months is really odd. I wonder if it's a capacitor failure in your wall control panel which is causing your wall panel to brown out & reboot. What model is your wall control panel?

Chamberlain LiftMaster Residential Jackshaft Opener Model 3800. The opener is powered through the battery backup which plugs into the wall. Could I still be having brown out? I don't have the ratgdo board on a UPS however.

neldonado, 2024-07-30:
> It only happening after months is really odd. I wonder if it's a capacitor failure in your wall control panel which is causing your wall panel to brown out & reboot. What model is your wall control panel?

Chamberlain LiftMaster Residential Jackshaft Opener Model 3800. The opener is powered through the battery backup which plugs into the wall. So I can shut off the breaker and the garage door functions fine.

neldonado, 2024-08-01:
> It only happening after months is really odd. I wonder if it's a capacitor failure in your wall control panel which is causing your wall panel to brown out & reboot. What model is your wall control panel?

I have a family member in the garage door business and they think it could just be due to age forgetting the rolling codes and spamming open / close. I am going to reset the controller and monitor this.

PaulWieland, 2024-08-01:
Your GDO doesn't use rolling codes for wireline control

neldonado, 2024-08-01:
> Your GDO doesn't use rolling codes for wireline control

Right, what I am saying is I think this may be unrelated to the ratgdo board. The technician said when it opens and closes randomly it's a sign of the controller losing the rolling code of the wireless openers and just picking up signals from nearby controllers. Something you normally only see with older devices that haven't been serviced in a long time. He walked me through holding the purple learn button to reprogram everything. If it happens after that then maybe it does have something to do with the ratgdo? unless you think otherwise.

sile16, 2024-08-24:
Also, experienced 2 of the 3 doors became unresponsive.   Here is the logbook from home assistant and also the mqqt status.    After a simple power cycle of the device it starts working again.

ratgdo v2.57 for hardware v2.50 , mqqt version
Security + 1.0
Chamberlain Liftmaster 

[image]

door status stuck in "sycing"
[image]

thenakuls, 2024-09-09:
I have secplus 1 GDO  889LM control panel and I have noticed that after a power blip (very short power outage causing garage door to restart) ratgdo stops working, most probably because it goes in emulation mode.

@PaulWieland would adding an option of adding a delay after reboot help?

neldonado, 2024-09-12:
@PaulWieland just to keep you in the loop on odd behavior I've been having. I had some power flickers last night, nothing shut off (to my knowledge no UPS alerts on my other equipment), but I'm wondering if it was enough to mess up the ratgdo. This morning I opened my garage door and the status was stuck on "Closed". Logging into the web server for it I was able to remotely reboot the device, and all worked fine after that. I'm wondering if simply getting a small UPS would fix these issues I'm having.

neldonado, 2024-09-26:
Again had my door randomly open / close last night. [image]

It didn't actually open on its own until 6:30am (not even captured in the logs, caught on my camera)

PaulWieland, 2024-09-26:
See my comment above.

neldonado, 2024-09-26:
[image]

It's just a few seconds between "was opened, was closed"

PaulWieland, 2024-09-26:
Have you tried unhooking your wall panel or replacing it?

neldonado, 2024-09-26:
> it?

Are you talking about this? If so, no, but I will!

[image]

closed 2024-10-09 by PaulWieland