ratgdo/mqtt-ratgdo #96: Why does sending a query command set all states to "unknown"

xelwarto, 2024-09-28:
Looking to understand why sending a **query** command initially sets all the states to "unknown":

```
if(command == "query"){
	Serial.println("MQTT: query");

	// Set all to unknown
	doorState = 0;
	lightState = 2;
	lockState = 2;
	motionState = 0;
	obstructionState = 2;

	if(controlProtocol == "secplus2"){
		getRollingCode("reboot2");
		transmit(txSP2RollingCode,SECPLUS2_CODE_LEN);
		delay(100);
	}
}
```

This results in the **statusUpdateLoop** sending a status message with an "unknown" state; once the state is updated, another status message is sent with the correct state.

```
QUEUE MSG ARRIVED [homegrg/gdohome/command] 
query
MQTT: query
rolling code for XXXXXXX 217|reboot2 : XXXXXXXXXXXXXXXXXX
Door state unknown
QUEUE MSG SENT [homegrg/gdohome/status/door] 
unknown
Light state unknown
QUEUE MSG SENT [homegrg/gdohome/status/light] 
unknown
Lock state unknown
QUEUE MSG SENT [homegrg/gdohome/status/lock] 
unknown
Obstruction status unknown
QUEUE MSG SENT [homegrg/gdohome/status/obstruction] 
unknown
Obstruction status clear
QUEUE MSG SENT [homegrg/gdohome/status/obstruction] 
clear
XXXXXXXXXXXXXXXXXX
XXXXXXXXXXXXXXXXXX | STATUS: door:2 light:1 lock:0 obs:1
Door state closed
QUEUE MSG SENT [homegrg/gdohome/status/door] 
closed
Light state on
QUEUE MSG SENT [homegrg/gdohome/status/light] 
on
Lock state unlocked
QUEUE MSG SENT [homegrg/gdohome/status/lock] 
unlocked
```

Is there anything in the documentation which talks about this? It would appear any code would have to ignore messages which return with an "unknown" state; this just seems unnecessary.

PaulWieland, 2024-09-28:
It sets it to unknown so you have positive confirmation that the real status was returned by the GDO itself. Otherwise there is no way to confirm the GDO actually responded.

closed 2024-09-28 by PaulWieland

xelwarto, 2024-09-28:
@PaulWieland  - I understand the need to know if the GDO actually responded; is there a reason why that is not part of the **gdoStateLoop** function?

xelwarto, 2024-09-28:
Without sending a query command, is there a way to determine if the GDO did not respond?

PaulWieland, 2024-09-28:
When you send a query command, the states go to unknown. If they remain unknown, the door didn't respond. If they change to a known state, the door did respond.