ratgdo/mqtt-ratgdo #97: command:query does not appear to work

jeremy44, 2024-11-05:
When I sent a MQTT message to my RATGDO [ratgdo v2.59 for hardware v2.50 Security 2.0] nothing subsequently happens, I'm expecting a message to be published to '[Device Name]/status/door'.  I'm following the directions published here: 

https://paulwieland.github.io/ratgdo/01_features.html

That is, publishing a query message to a topic '[Device Name]/command' with the body of 'query'. 

Thanks in advance.

michaeljoos72, 2024-11-12:
For me '[Device Name]/command/command:query' does work --> 2x command

[image]

jeremy44, 2024-11-12:
@michaeljoos72 Thanks so much.  That worked. That wasn't clear to me from the documentation or the code.  Appreciate it.

closed 2024-11-12 by jeremy44